ArduPilot/ardupilot #26815: SBUS Output: Channel 5 also being output on Channel 13

rmackay9, 2024-04-16:
In Copter-4.5 (and higher) the SBus output feature ([perhaps best described on the wiki here](https://ardupilot.org/copter/docs/common-sbus-out.html)) is outputting the Channel 5 value to Channel 13 as well.  There are likely other channels that are being output incorrectly as well.

This bug was reproduced on two daisy-chained CubeOrangePlus autopilots by doing this:

1. Loaded 4.6.0-dev onto both autopilots, wiped parameters, performed accel, compass and RC calibration
2. 1st autopilot's parameters were set:
    - BRD_SBUS_OUT = 3 (100hz)
    - SERVO5_FUNCTION = (RCIN6)
3. 1st autopilot's SBus out was connected to 2nd autopilot's RCIN port
4. 1st autopilot's RC input channel 6 was connected to a received an a transmitter's tuning knob was used to change the RC input
[image]
5. When 1st Copter running Copter-4.4.4 the 2nd autopilot's RC input correctly shows only channel 5 changing
    [image]
6. When 1st Copter running Copter-4.6.0-dev the 2nd autopilot's RC input shows both channel 5 and 13 changing
    [image]

This bug was first reported [here on the Copter-4.5 forum](https://discuss.ardupilot.org/t/copter-4-5-0-released/115614/21)

andyp1per, 2024-04-16:
I would think that the most likely candidate as a cause for this are the iomcu changes

tridge, 2024-04-17:
this is worth more investigation, a Saleae trace on the sbus out with 4.5.1 would be good, see what the bits actually look like

closed 2024-04-17 by tridge

tridge, 2024-04-30:
@rmackay9 the duplication of the lower 8 channels to upper 8 channels is just a read of the reg_direct.pwm values which are next in line in ram. It will not cause a crash, and it will always be that with this bug the upper 8 channels are a copy of the lower 8 channels