ArduPilot/ardupilot #26816: AP_HAL_ChibiOS: disable safety switch on more boards

rmackay9, 2024-04-16:
We should disable the safety switch on at least some boards

- Holybro autopilots
- CubeRedPrimary/Secondary

I think most manufacturers are moving away from safety switches but we need to be careful about the transition to be sure that users are not surprised.  We should only make this change as part of a major release where there are more expectations for change.

tridge, 2024-04-22:
we can change BOARD_SAFETY_ENABLE_DEFAULT to 0 for these boards

rmackay9, 2024-04-22:
We decided to only change the safety switch for Holybro boards for 4.6 (and higher) and see how that goes and perhaps in 4.7 we will do it for all boards if that's what the autopilot vendors would like.

We also discussed that it would be nice if the Mission Planner had Safety Switch ON and OFF buttons (currently there is a toggle but it's hidden on the Ctrl-F screen).  This software level safety switch feature is useful for some high end users (mostly planes and quadplanes).