ArduPilot/ardupilot #26824: Small error in calculate_yaw_and_rate_yaw() in AC_PosControl?

ashtorak, 2024-04-16:
Looking at this line:

https://github.com/ArduPilot/ardupilot/blob/7f61acedd3f973af9d9febfe6d76e71d12ae9cde/libraries/AC_AttitudeControl/AC_PosControl.cpp#L1252

Is it supposed to be like this? It doesn't make much sense to me though. I would have expected something like

degrees(turn_rate) * 100

to convert it to centidegrees, even though I have to admit that I haven't understood yet how the turn rate is calculated in the first place, but degrees(some decimal based float) seems very wrong in any case ;)

IamPete1, 2024-04-16:
The degrees function multiplies the passed in value by `180.0/pi`. If we were to expand that out we would get:
```
turn_rate * 100.0 * (180.0/pi)
```
The order of multiplications does not matter so
```
turn_rate * degrees(100.0)
```
is the same as
```
degrees(turn_rate) * 100
```

However I do agree that your suggestion is easier to understand.

closed 2024-04-16 by IamPete1

ashtorak, 2024-04-16:
True, the end result is the same. Just a little confusing when you see it like this.