ArduPilot/ardupilot #26860: Vagrant Working on Arm Machines

jonathan-k4, 2024-04-21:
## Feature request

It would be great if vagrant worked on newer arm based machines (Apple silicon, etc) as Virtualbox still doesn't have an arm64 release. 

I was able to get it to work with Docker by doing the following. Would offer up a PR, but not sure how it would need to be integrated to keep the community happy.

* Edit `Vagrantfile` and add QEMU compatible box above `ubuntu/jammy64` block:

```
  # 22.04 Generic Ubuntu with QEMU support for ARM macOS
  config.vm.define "ubuntu64", primary: true do | ubuntu64 |
    ubuntu64.vm.box = "tknerr/baseimage-ubuntu-22.04"
    ubuntu64.vm.provision :shell, path: "Tools/vagrant/initvagrant.sh"
    ubuntu64.vm.provider "docker" do |vb|
      vb.image = "tknerr/baseimage-ubuntu:22.04"
      vb.has_ssh = true
      vb.name = "ArduPilot"
    end
    ubuntu64.vm.boot_timeout = 1200
  end
```

and change `ubuntu/jammy64` to not be primary:

```
  # 22.04 LTS EOL Apr 2032
  #config.vm.define "jammy", primary: true do |jammy|
  config.vm.define "jammy", autostart: false do |jammy|
```

* Edit `Tools/vagrant/initvagrant.sh`, comment out resize, and add tzdata command to avoid prompt caused in later script:

```
#sudo resize2fs /dev/sda1
sudo apt-get update
DEBIAN_FRONTEND=noninteractive TZ=Etc/UTC sudo apt-get -y install tzdata
```

and comment out ptrace lines:

```
#perl -pe 's/kernel.yama.ptrace_scope = ./kernel.yama.ptrace_scope = 0/' -i /etc/sysctl.d/10-ptrace.conf
#echo 0 > /proc/sys/kernel/yama/ptrace_scope
```

* Start Vagrant

```
vagrant up --provider qemu
```

* SSH into vagrant

```
vagrant ssh ubuntu64
```

* Initialize submodules

```
cd /vagrant
git submodule update --init --recursive
exit
```
* Run ArduPilot in SITL mode

```
vagrant ssh ubuntu64 -c "sim_vehicle.py -j 2 -v Rover"
```

jonathan-k4, 2024-04-21:
Maybe this is just an extra page in the manual, not a PR?

peterbarker, 2024-05-07:
I wouldn't mind a PR, but I think the changes could be a little less intrusive.  See my PR for adding Noble support (https://github.com/ArduPilot/ardupilot/pull/26985) for an example of where the resize step is skipped.

Basically after maximum compatability / not breaking existing behaviour.  So changing `jammy` to not be primary is problematic as the vast majority of users will want that one.

... also, why remove the ptrace lines?

peterbarker, 2024-07-10:
@jonathan-k4 would be nice to get this in.  I can possibly put together a PR based on your issue here, but I don't have any means of testing that it works on arm64.

peterbarker, 2025-05-11:
Ping @jonathan-k4 still interested in this one?

jonathan-k4, 2025-05-11:
Sure, I can test it if you have a PR.