ArduPilot/ardupilot #26863: Control of the Alexmos 32-bit gimbal from via RC does not work.

hyperion11, 2024-04-22:
Control of the Alexmos 32-bit gimbal from via RC does not work. If i roll back to version 4.4.4 then everything works. From the Mission Planner , gimbal control works in both 4.4.4 and 4.5.1. The gimbal is configured via the Mavlink v2 protocol.

[image]

Plane 4.5.1
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Fixed Wing

**Hardware type**
MFE Pixpilot V3

timtuxworth, 2024-04-22:
I'm seeing something similar with the MFW PixPilot V3 with a servo gimbal. I have just started to research so I don't have all the data yet, but I saw this post so I thought I'd add what I know. The gimbal mount may not be working with ArduPilot 4.5.1 on this board.
I'm wondering if it's a memory issue, there is not much memory left, hovers between 3k and 7k on this board.

hyperion11, 2024-04-22:
I tried to compile the firmware with a minimum set of modules. It did not help.

hyperion11, 2024-04-22:
also on this tab i see buf overflow above 40k on 4.5.1. And about 4k on 4.4.4.The reason may be the serial port communication.
[image]
But AHRS, GPS, RC, and CONTROL status OK in simplebgc GUI.

timtuxworth, 2024-04-22:
Interesting I'm seeing the gimbal control not working and also MSP DisplayPort doesn't seem to be working - which is a serial port connected to my HD-Zero VTX.

closed 2024-04-22 by rmackay9