ArduPilot/ardupilot #26877: Rover/Boat: improve steering for mixed Ackermann+SkidSteering vehicles

rmackay9, 2024-04-24:
In cases where a Rover or Boat uses both Ackmerann and Skid-steering ([see types here](https://ardupilot.org/rover/docs/rover-motor-and-servo-configuration.html)) users find that the skid-steering motors adjust their throttle too much especially at higher speeds.

We should be able to provide a smarter blending of steering output where skid-steering provides all the turning while at low speeds and the rudder provides all the steering at very high speeds.

This request comes from [this Rover-4.5 discussion](https://discuss.ardupilot.org/t/gain-for-throttle-l-and-r/116735).

geofrancis, 2024-04-24:
I had something but i never got around to testing it, basically use a lua to reassign outputs and pid on the fly, It kind of worked but could only switch when disarmed.

 what about something like propulsion profiles?  on my boat i was going to have 3 modes, one with just rudder and propulsion, one with rudder and skid steer and the third with thrusters. so what about a file on the sd card with a list of servo out values and pid profiels that can be switched using an aux function..

xianglunkai, 2024-04-25:
Perhaps it is possible to establish this type of dynamic distribution equation, linearize it, and optimize it online based on some indicators to obtain the control distribution matrix