ArduPilot/ardupilot #26892: AP_Mount: mount_open servo output should work for all backends

rmackay9, 2024-04-26:
The SRV_Channel library has a [mount_open](https://github.com/ArduPilot/ardupilot/blob/master/libraries/SRV_Channel/SRV_Channel.h#L58) and mount2_open functions but they are only used by the [Mount_Servo backend](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount_Servo.cpp#L112).

We should move the little bit of code that controls to servo into a function in [AP_Mount_Backend](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount_Backend.h) and then call it from every backend as part of the update() call.

videh25, 2025-01-10:
Hey @rmackay9 
I am new to ardupilot codebase but want to contribute for exploring open source development. 
I have spent some time trying to understand this issue and the related code base. 

**Can you please tell me if my understanding is correct?**
1. AP_Mount is a library written for controlling the camera mounts.
2. Right now the functionality mount_open, which is used for deploying or retracting a camera mount has been implemented only for Mount Servo driver.
3. However, even the drivers of other mounts can use servo based communication to control the deployment and retract the mounts.
4. Thus the issue is for extending the functionality to all mount drivers.

**Expected Solution (Please correct me if I am wrong)**
1. Move the servo mount functionality to AP_Mount_Backend, the parent class of all mount drivers so that every mount driver has a servo based implementation of deployment and retraction of camera mount.
2. And call the open_mount control function in every AP_Mount_Backend's child class's update call. From every class, I mean all the currently implemented classes, like AP_Mount_Alexmos,  AP_Mount_CADDX and others.

I have also been through the PR raised by @rohan-pandeyy and have also understood the issues raised in that conversation. Can I please work on this issue, if no one else is working on it?

rmackay9, 2025-01-10:
Hi @videh25,

Yes, your understanding looks correct to me so go for it.  I don't think anyone else is working on it at the moment

videh25, 2025-01-15:
Hey @rmackay9,
I have made some changes and created the PR. Please let me know if I should do anything else to solve the issue. 

Glad to be able to contribute!

videh25, 2025-01-15:
I also realised that my changes are different from the ones done by @rohan-pandeyy in his PR. 

I realise that he extracted the fetched the servo channels and directly assigned the PWM functions to them. I have tried to set PWM values via the higher level functions, as done in the original code. Please correct me if I am missing something.

amilcarlucas, 2025-10-27:
Just do a github.com PR. No need to assign this to you.

rmackay9, 2026-03-18:
@hunt0r yes, it was, thanks!

closed 2026-03-18 by rmackay9