ArduPilot/ardupilot #26900: AP_LOGGER: flash nand incorrect behaviour

BloodSakura3774, 2024-04-27:
## Bug report
Tested on W25N01 with our custom board and original driver, W25N02 with our driver, and W25Q128 with Foxeer H743 and original driver. 
After a full write (Until GCS says "chip full, logging stopped"), the log downloaded has no FMT data and can not be opened with MP's log viewer. And after a reboot of the FC, it will erase itself, which I believe should not happen according to wiki (https://ardupilot.org/copter/docs/common-downloading-and-analyzing-data-logs-in-mission-planner.html)
Then after another manual erase, the GCS would say "No last page of log X at top=Y or bot=Y" when opening log download page but the log is indeed continuing to be written.

**Full Procedure to make it happen**
1. Enable LOG_DISARMED, set LOG_BITMASK to 3588093, set INS_RAW_LOG_OPT to 10, set INS_LOG_BAT_OPT to 5, set INS_LOG_BAT_MASK to 3, and other remains to default. (I don't think these are required but I set them to speed up the test)
2. Reboot and Manual erase the chip.
3. Wait until GCS says "chip full, logging stopped"
4. Download the log and open it, it should say No FMT data.
5. Reboot and the previous log should already be erased by itself and then do another Manual erase.
6. After erase complete message, open the log downloading page and there should be a very small log (246 on W25Q128 and 20xx on W25N01/02) and GCS message tab says "No last page of log X at top=Y or bot=Y", with X stays the same the Y changing every time you open the download page.
[image]
[image]

7. Do another reboot before GCS says "chip full, logging stopped", and the log generated from step 6 should appear and downloading it and opening it should have no error.
**Version**
At least on ArduCopter V4.6.0-dev (338f492a)
Not tested with prior version

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine
Only tested with copter firmware but I believe it will happen to other platforms since its logger related.
**Airframe type**
N/A

**Hardware type**
Custom board with STM32H743 and Foxeer H743
Should happen with and FC that uses W25N01/W25N02/W25Q128 flash nand chip (At least this three nand)
**Logs**
N/A

andyp1per, 2024-04-28:
So a big part of what you are seeing here is a result of logging too much data. The FMT messages are sent out on arming but can be severely delayed if there is a lot of logging going on. This is not a problem with the driver per se but the speed of the device. One solution is to to set a logging rate that matches the speed of the chip.

BloodSakura3774, 2024-04-29:
Tested with log_bitmask set to only log medium atti. And used gcs to force arm the drone to start logging, the problem persists. First log still says no FMT and after reboot, erase, then the second log is not downloadable. It then disappears  after another reboot.

peterbarker, 2024-07-10:
@BloodSakura3774 could you retry with master, please?  I've pushed through a fairly sweeping change of the way we get formats out to files.

This *feels* like a problem somewhere lower, however.

If master hasn't fixed the problems I can look at ordering a FoxeerH743 to try to replicate locally.

BloodSakura3774, 2024-07-10:
> @BloodSakura3774 could you retry with master, please? I've pushed through a fairly sweeping change of the way we get formats out to files.
> 
> This _feels_ like a problem somewhere lower, however.
> 
> If master hasn't fixed the problems I can look at ordering a FoxeerH743 to try to replicate locally.

My Foxeer H743 is under use, but I'll take a look with the custom one (with w25n02/01)

BloodSakura3774, 2024-07-11:
@peterbarker This issues persists.
Step 4, after download the log still shows no FMT data and it also tells me download finished in the middle, like 10mb downloaded or 50, 120mb downloaded.
Step 5, it still shows "No last page of log X at top=Y or bot=Y" and a very small log. 

Tested with the custom fc with w25n02, didn't test w25n01 or w25q128/256