ArduPilot/ardupilot #26926: SITL: SIM_GPS_DISABLE should work when AHRS_EKF_TYPE = 10

rmackay9, 2024-04-30:
When AHRS_EKF_TYPE = 10 (SITL) the SIM_GPS_DISABLE parameter doesn't have the expected result (which is that the EKF should fail to produce a position estimate).

peterbarker, 2024-05-07:
The "perfect" EKF type, 10, doesn't rely on sensors for data, it is just echoing back the simulation numbers.  There's going to be a lot of things we'll ignore - all of the simulated sensor "dirtying" parameters, for example.  SIM_GPS_DISABLE is just the most obvious one.

We could add a set of flags to allow for changes in EKF-type-10's output?

rmackay9, 2024-05-07:
Thanks @peterbarker.  I agree this could be a very large tasks so we should consider if it is worth the effort or if the user who reported this can perhaps find another solution instead.

Muteeh05, 2024-11-20:
Is there any way I can turn on SIM_GPS_DISABLE parameter with AHRS_EKF_TYPE=10?

peterbarker, 2024-11-24:
@Muteeh05 as stated above, there's no GPS to disable - the simulated AHRS gets perfect location.

You can use RC channel 65 to force the GPS fix-type to 1 which does a lot of what people want from this.

Closing this one now.

closed 2024-11-24 by peterbarker

rmackay9, 2024-11-24:
Hi @peterbarker, I think this is a valid enhancement request actually.