ArduPilot/ardupilot #26943: Allow any serial port to be used for RC input, while detecting the type of protocol

mvrius, 2024-05-01:
## Feature request

**Is your feature request related to a problem? Please describe.**

I have a Navio2 RPI Hat. While trying to set up Remote Control for the hat, I tried connecting my ELRS receiver, outputing either CRSF or SBUS directly to the UART port of the Navio2.

Using the CMD line arguments I passed in the serial port data from UART to Serial 5. My expectation was that configuring RC to protocol detect + Serial5 to RCIN, would mean that Serial5 input would be used for RCIN, then protocol detection kicks in.

In reality that doesn't seem to happen. The only input considered for RCIN is the actual RCIn Driver provided for Navio2 board.

**Describe the solution you'd like**

Allow any serial port to be used for RC input, while detecting the type of protocol. 

**Describe alternatives you've considered**

I've unblocked myself by getting a PWM to PPM convertor, but I feel this should be solved in software.

**Platform**
[ ] All
[ ] AntennaTracker
[X] Copter
[X] Plane
[X] Rover
[X] Submarine

**Additional context**

Willing to contributed changes to this if somebody can help with some guidance on how this should be implemented. My analysis so far:

Looking at other HAL examples, ESP32 initializes the AP::RC in the RCInput Driver. But this doesn't seem to work as expected on Navio2 because AP_RCProtocol should be initialized after AP_SerialManager. Logging steps on Navio2 shows that find_serial of SerialManager will not find the configured RCIN serial_port.

There seems to be some wiring missing to allow for Serial RCIN to be supported on all types of UARTs and platforms.

kalzoo, 2024-05-03:
Drive-by comment: have you tried sending SBUS to the PPM/SB pins at the end of the servo rail? Is that viable in your build?

mvrius, 2024-05-03:
@kalzoo - yes, that works as expected. Build wise I'm not blocked.

Allowing UART/any serial though should work or fail explicitly, so it's clear that RCIN configuration for serial is not support on certain boards.

I think it may also open future avenues, since configuring the protocol as CRSF may allow to send back telemetry to RC transmitters.

peterbarker, 2024-05-07:
As luck would have it I was just starting to look at fixing the Linux HAL in terms of RC input so that everything goes via the AP_RCProtocol library.

I don't *think* this is going to work as-is for getting RC input from the NavIO2's UART, there's going to be inversion problems at a guess.  I do have a receiver wired up but haven't tried to configure things yet,

Closed by https://github.com/ArduPilot/ardupilot/pull/27000

mvrius, 2024-05-07:
@peterbarker this is great! 

Took a quick look at the PR. I did similar changes in my attempt to enable any serial to be used on Navio2, and it seemed to struggle with protocol detection. I think your intuition is right that there is going to be an inversion of problems.

If you need testers for this, I can gladly test this branch in June when I return back home from vacation.

huntingcodes-001, 2024-09-22:
Heyy I can solve it, can you please assign it to me? @kalzoo @peterbarker