ArduPilot/ardupilot #26968: Can implement Custom Control in Plane

kmuthugtk, 2024-05-03:
## Feature request
Can I implement the Custom Control in Plane and if can. Could you provide some documents/any repository.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[**] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

magicrub, 2024-05-03:
It's open source so you can do anything you want. Can you be more specific?

kmuthugtk, 2024-05-06:
AC_CustomControl has been added in the Arducopter. In the similar fashion do i add the same AC_CustomControl to the Arduplane.If it is Can you give a brief steps to follow how do i add the custom controller to the Arduplane.

peterbarker, 2024-07-10:
> AC_CustomControl has been added in the Arducopter. In the similar fashion do i add the same AC_CustomControl to the Arduplane.If it is Can you give a brief steps to follow how do i add the custom controller to the Arduplane.

We don't have the same infrastructure as Copter to implement custom controllers directly into Plane.

However, you might have a look at the Aerobatics LUA script here: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/applets/Aerobatics/FixedWing/plane_aerobatics.lua

That essentially does *all* of its own control, at some levels at least.

I'm afraid this isn't really an "Issue" as currently written, so I'm going to close this.

closed 2024-07-10 by peterbarker