ArduPilot/ardupilot #26980: DFU / USB Device Descriptor broken in Bootloader

menschel, 2024-05-05:
## Bug report

When trying to enter DFU mode via boot button, the device fails to enumerate correctly and cannot be flashed by STM cube programmer for example.

**Version**
stable 4.5
https://firmware.ardupilot.org/Copter/stable/KakuteF4/

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Diatone Tyrant S 2017 (Quad + BetaFlightX)

**Hardware type**
Holybro Kakute F4 V2.4 , came with Betaflight pre-installed, flashed to Ardupilot via DFU. Cannot flash back to Betaflight now.

**Logs**
[ 1032.069068] usb 5-4: new full-speed USB device number 6 using xhci_hcd
[ 1047.524802] usb 5-4: device descriptor read/64, error -110

andyp1per, 2024-05-05:
DFU is built into the hardware. Nothing to do with ardupilot.

closed 2024-05-05 by andyp1per

menschel, 2024-05-06:
Hello again,
it seems I was mistaken and it is a hw issue caused by having a ELRS Receiver connected to R3,T3,5V,GND.
The device enumerated after cutting the wires.