ArduPilot/ardupilot #26993: Takeoff in GUIDED mode without GPS/GNSS but SLAM

engcang, 2024-05-06:
## Feature request

Hi. I have been used PX4 with Pixhawk for a long time implementing many autonomous flying drones with SLAM or VIOs (you may want to refer to the clips [1](https://youtu.be/bX2ZsTqsRfY?si=cLSoK_1IiujC9vHI), [2](https://youtu.be/ft0okSqnyts?si=Dw91IZ-S0ro8zQ9T), [3](https://youtu.be/Ux6ir2_ayoY?si=Rf0HuMuUyyBrEnq_), etc.
I recently had some chanced to try Ardupilot and found it great to tune controller gains.
I successfully and easily migrated my former autonomous flying algorithms and tested them with the same code except the takeoff and landing. I manually tookoff the drone and then changed the mode to GUIDED.

Sadly, I Googled as much as possible and pretty sure that takeoff in GUIDED mode without GPS/GNSS but SLAM, without manually clicking "Set home" on the MissionPlanner, without manually taking off the drone and then change the mode to GUIDED to fly the drone autonomously.

I kindly ask if there exists the proper way to takeoff the drone in GUIDED mode without any human actions but only codes, without GPS/GNSS but SLAM (I am using /mavros/vision_pose/pose currently).

Otherwise, I would like to ask if you have any plans to implement the functional in the near future, or why you would not.

**Describe the solution you'd like**

1. Takeoff command supported in GUIDED mode
2. Otherwise, waypoint command in GUIDED mode takesoff the drone after arming.

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2024-05-07:
Hi @engcang,

In general users should set the EKF origin using the ground station. E.g. open MP's Flight Data screen and right-mouse-button click on the map and select Set Home Here, Set EKF Origin Here

We've got a [Lua script here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/examples/ahrs-set-origin.lua) that could be modified and used here.

I've created a wiki page issue here: https://github.com/ArduPilot/ardupilot_wiki/issues/5935

I'll go ahead and close this issue but I'll see replies in case you think the above isn't sufficient.

closed 2024-05-07 by rmackay9

engcang, 2024-05-07:
Hi @rmackay9 

After I publish the SLAM's estimated pose via /mavros/vision_pose/pose, which is VISION_POSITION_ESTIMATE, mavproxy outputs
```bash
AP: EKF3 IMU0 yaw aligned
AP: EKF3 IMU1 yaw aligned
AP: EKF3 IMU0 is using external nav data
AP: EKF3 IMU0 initial pos NED = 0.0,-0.0,-0.1 (m)
AP: EKF3 IMU1 is using external nav data
AP: EKF3 IMU1 initial pos NED = 0.0,-0.0,-0.1 (m)
```

and also mavros which is connected to the pixhawk outputs
```bash
[ INFO] [1715073248.269162075, 82.965000000]: FCU: EKF3 IMU0 yaw aligned
[ INFO] [1715073248.269425090, 82.965000000]: FCU: EKF3 IMU1 yaw aligned
[ INFO] [1715073248.271926815, 82.968000000]: FCU: EKF3 IMU0 is using external nav data
[ INFO] [1715073248.272076282, 82.968000000]: FCU: EKF3 IMU0 initial pos NED = 0.0,-0.0,-0.1 (m)
[ INFO] [1715073248.272214568, 82.968000000]: FCU: EKF3 IMU1 is using external nav data
[ INFO] [1715073248.272355966, 82.968000000]: FCU: EKF3 IMU1 initial pos NED = 0.0,-0.0,-0.1 (m)
```

1. Are not these the outputs of setting the EKF origin you mentioned?
(FYI, as I wrote in the first comment, after manually taking off the drone and then change the mode into GUIDED, the drone flies well autonomously. So the EKF parameters are set properly.)
2. From somewhere I found that the drone must be armed after the mode is changed to GUIDED, otherwise the drone cannot takeoff in GUIDED mode. Is this correct information?
3. I found that setting home and setting origin for EKF/AHRS are different at here - https://ardupilot.org/dev/docs/mavlink-get-set-home-and-origin.html. What exactly do we have to set to takeoff in GUIDED mode?

rmackay9, 2024-05-07:
Hi @engcang 

1. sending in the current positions to the EKF from an external system is not the same as setting the EKF origin.
2. the vehicle must be armed before the motors can spin so it won't fly unless it is armed
3. yes, the EKF origin and AHRS home are two different things.  Normally home is initialised automatically to the vehicle's current location soon after the EKF origin is set.  So normally only the EKF origin needs to be set.  Setting Home is not required.

Perhaps I'm misunderstanding some questions so apologies if you already know some of the above.

Hope that helps.

FahrulID, 2024-05-08:
@engcang 

In Mavlink To set the EKF Origin, you need to send SET_GPS_GLOBAL_ORIGIN, in Mavros you can use global_position plugin with this topic `/mavros/global_position/set_gp_origin`. The topic used [geographic_msgs::GeoPointStamped](http://docs.ros.org/en/noetic/api/geographic_msgs/html/msg/GeoPoseStamped.html) message.

engcang, 2024-05-08:
@rmackay9 
Thank you for your detailed and fast reply. 
With the Lua script you attached, the drone can successfully take off in GUIDED mode without GPS/GNSS.
The second question was about the order of command. I read somewhere that arming -> mode change into GUIDED cannot takeoff drone, but only mode change into GUIDED -> arming -> then drone can takeoff. Is this true information?

I was about to ask how I can set the EKF origin without Lua script but with mavros (mavlink) command, but thanks to @FahrulID  I just checked it works well!

I've got an another question about landing in GUIDED mode.
In PX4, with OFFBOARD mode (GUIDED in Ardupilot), the command with the z-axis downward flies the drone vertically downward and then it disarms automatically after it contacts the ground (i.e., mavros/setpoint_position/local's translation values).

So, is it same in Ardupilot? or do I have to use /mavros/cmd/land again to land the drone as we have to do for the takeoff?

rmackay9, 2024-05-08:
Hi @engcang,

Yes, we've got a wiki page here which describes the commands to control the vehicle (in Guided mode) and includes the Takeoff command. https://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html

FahrulID, 2024-06-11:
@engcang sorry to bring this issue up again, could you please tell me your setup, as I have issue on VISION_POSITION_ESTIMATE using Realsense T265 where my LOCAL_POSITION_NED overshoot so much before going to the right position.

engcang, 2024-06-11:
@FahrulID Try to set VISO_DELAY_MS higher. I use 100ms.

rmackay9, 2024-06-12:
Hi @FahrulID,

The best way to get help is probably to post in the [support forums](https://discuss.ardupilot.org/c/arducopter/6) and include an onboard log.  As we say, it's all guesswork without a log.  The issue could be either an estimation or a control issue.