ArduPilot/ardupilot #27017: Ardupilot v4.5.1 Batt4 Percentage Issue

WaqasRauf, 2024-05-08:
## Bug report

**Issue details**

I've updated arduplane from v4.4.4 to v4.5.1, after the update my reventec fuel level sensor is not giving fuel percentage, I've checked voltages they are accurate but somehow ardupilot is unable to translate these voltages to percentage. My fuel sensor is connected on batt4. I've set the parameters as described in ardupilot wiki, I've checked the integration again by downgrading the version to 4.4.4 it's working properly but on 4.5.1 it's percentage is not accurate it's giving -1 in the batt4 percentage parameter.

**Version**
Arduplane v4.5.1

**Platform**
Plane

**Airframe type**
Quad plane (X)

**Hardware type**
CUAV X7+ Pro

**Logs**

IamPete1, 2024-05-08:
Please can you provide a log or parameter set. Which physical pin is the sensor connected to?

WaqasRauf, 2024-05-15:
Dear Peter,
Please find the attached dataflash log file. The Reventec Fuel Level Sensor
(LS300-225) is connected on Pin 18 and it is configured on BAT4, You can
see in the logs battery voltages are there but there is no percentage.

Regards,
Waqas
 00000044.BIN
<https://drive.google.com/file/d/1K88H4qcuiDuArMT89GMEEWjSQHSF0sY4/view?usp=drive_web>

IamPete1, 2024-05-19:
Fix here: https://github.com/ArduPilot/ardupilot/pull/27099

closed 2024-05-21 by tridge