ArduPilot/ardupilot #27031: GSOF: Supply 5Hz GPS for robust EKF

Ryanf55, 2024-05-09:
## Feature request

**Is your feature request related to a problem? Please describe.**

PX-1 provides fused GPS to EKF3 right now, which couples IMU.

**Describe the solution you'd like**

Add support for a new raw GPS message. 
We want position, velocity, and 1 sigma accuracy for both pos/vel. Velocity should include horizontal and vertical accuracy.

**Describe alternatives you've considered**

* Coupling the EKF's

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**

N/A

moreba1, 2025-01-30:
Hi
Isn't 5 HZ RATE GPS used for EKF now?

Ryanf55, 2025-01-31:
The PX-1 can do 100Hz, but we send it right now at 5Hz.