ArduPilot/ardupilot #27064: Ardupilot V4.5.x - ESC telemetry on the OSD has stopped working

robustini, 2024-05-14:
From Ardupilot V4.5.x you can also use ESCs with telemetry (e.g. BLHeli) from SERVO1 to SERVO8, whereas with earlier versions you would have to use SERVO9 and up, so currently everything is smarter.
Unfortunately, for those like me who use ESC telemetry with OSD and had therefore utiulized connectivity from SERVO9 on up now it does not work.
Previously connecting an ESC to SERVO9 was interpreted as “ESC1” and the data on the OSD was interpreted, now it is seen as “ESC9” and the OSD does not work perhaps because it precisely goes looking for it on SERVO1.
It is in my opinion a bug that should be fixed, simply by mapping on the OSD as “ESC Telemetry” the first ESC connected in ascending order, even if for example it is wired to SERVO9.

andyp1per, 2024-05-15:
This is a pre-existing bug. The OSD ESC information would only show info about the "first" ESC, including ESCs that did not exist. So unless you physically had your drone setup in a particular way you would not get ESC information. The change in 4.5 is that the "first" ESC is now SERVO1 rather than AUX1 due to the support for ESC telemetry on IOMCU. The solution is to either allow the ESC to be configured to to automatically determine a reasonable ESC.

robustini, 2024-05-15:
Absolutely agree, just be able to configure in the OSD which ESC to show as ESCTEMP, ESCRPM and ESCAMPS (I never understood why ESCVOLTS is absent, telemetry voltage is in virtually all ESCs with telemetry).
Now you cannot so taking SERVO1 as reference if you have ESCs connected in other outputs the telemetry will not work.

robustini, 2024-05-18:
And today by doing a test I noticed that the telemetry problem also goes to create problems if for example you set the ESC to SERVO1, declaring it in the bitmask of BLHeli goes to disable the pwm of SERVO2.
So yes, it is a bug.
Tested on SpeedeBee F405 WING.

robustini, 2024-08-02:
Thanks @andyp1per!
https://github.com/ArduPilot/ardupilot/pull/27136

closed 2024-08-02 by robustini