ArduPilot/ardupilot #27078: ardu plane 4.5.2 flight modes not processed

ilioSS53, 2024-05-16:
**Issue details**
Flight modes not processed responding updated
_Please describe the problem_
Version 4.5.2 uploaded to my system.
Mission planner
I am using frsky x20s rx and frsky td xm rx
Speedybee f405 wing app
Former revision the fm where working from my function switches 6pcs
Stabelize, fbwa, fbwb, manual, training, rtl
After change to 4.5.2 fm not responde to function switch.
Only after reboot the selected fm will be shown on the datascreen
All work as should bee like servo input output fm on ch6 changed on pwm changes.

**Version**
_What version was the issue encountered with_ 4.5.2

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Glider
**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Speedybee f405 wing app
**Logs**
_Please provide a link to any relevant logs that show the issue_
Not yet

rmackay9, 2024-05-16:
Hi @ilioSS53,

Thanks for the report.  Can you raise this in the [Plane-4.5 forum](https://discuss.ardupilot.org/c/arduplane/plane-45/176)?  Maybe raise a new topic there.

I'm going to go ahead and close this because the regular procedure is that support items should first be raised on the forums and then once they're confirmed to be bugs or enhancement requests we raise them here.  Sorry to be strict but if we're not then our to-do list (which is essentially what this is) just turns into another support forum.

closed 2024-05-16 by rmackay9

ilioSS53, 2024-05-17:
Hi, Randy,

I will do.
Sorry to misunderstand the procudure.

Kind regards,
IliossS53

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>