ArduPilot/ardupilot #27102: MOT_PWM_MIN, MOT_PWM_MAX are reset to default values automatically

StenPetrov, 2024-05-20:
## Bug report

MOT_PWM_MIN, MOT_PWM_MAX are reset to default values automatically for arducopter.

**Repro:**
  - Open MissionPlanner
  - Connect to your copter
  - find MOT_PWM_MIN and MOT_PWM_MAX in the full parameter list
  - change their values (in my case MOT_PWM_MAX=1490)
  - Write Params
  - Disconnect copter, ensure it rebooted
  - Connect copter to MissionPlanner again, check the values of these settings - they are back to their default values

**Rationale**
This bug prevents the use of reversible ESC where half of the range is for reverse.
The goal is to be able to reverse the motors in script, but prevent the throttle from reversing the motors

**Version**
ArduCopter V4.5.2 (291be848)

**Platform**
[?] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Frame: QUAD/X

**Hardware type**
Pixhawk6X

rmackay9, 2024-05-20:
Copter doesn't support reversible motors though.  The most similar frame that supports reversible motors would be the heliquad frame using the TradHeli software.

IamPete1, 2024-05-20:
The work around would be to not set Dshot in `MOT_PWM_TYPE` and use the `SERVO_BLH_` params instead.

StenPetrov, 2024-05-20:
@rmackay9 HeliQuad is much further from my drone than a Quad. I doubt moving in that direction would be productive.
The ESCs are DShot indeed. BLHeli are all configurable in their range, including reversible, so enforcing a fixed value doesn't look right. I have a business reason to use reversible motors - when the drone is not in flight the motors can do other things; it would still fly like a quad.

I posted this here, because the customer expectation is to see things that work - if a particular setting doesn't do anything it should either be made to do its thing or be removed. Showing things that don't work is a bug.

@IamPete1 Can you please clarify your suggestion? What should I set MOT_PWM_TYPE to and which SERVO_BLH_ properties set to what values? In my setup MOT_PWM_TYPE is Dshot150 and below is a screenshot of my SERVO_BLH_ properties, including two of them than mention reversing

[image]

IamPete1, 2024-05-20:
`MOT_PWM_TYPE` should be 0. Sounds like you want 3D not reversed.

StenPetrov, 2024-05-20:
@IamPete1 What I ultimately need to achieve is the ability to move any set of motors in either direction while the drone is on the ground, via a Lua script. It would suffice if the servo range settings are actually used instead of hard-coded 1000-2000.

Setting MOT_PWM_TYPE to 0 doesn't change anything, the normal motor servo range still gets reset to 1000-2000 and the remote's throttle still maps to the whole range. Are there other settings you're suggesting I should change?

IamPete1, 2024-05-22:
I don't think this is a bug, it sounds like the @StenPetrov is trying to do something outside of what copter was designed for.

closed 2024-05-22 by IamPete1

StenPetrov, 2024-05-22:
@IamPete1 Fair - Then the bug is to remove the settings whenever they don't apply, so others like me don't waste their time - and yours - trying to figure out how to coax their copter into doing new things, or at the very least explain in the description boxes how it works - it's still a bug