ArduPilot/ardupilot #27157: Ubuntu 24.04 environment issue

Blake-West, 2024-05-25:
## Bug report

./install-prereqs-ubuntu.sh fails on Ubuntu 24.04 
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04 LTS
Release:	24.04
Codename:	noble

_Please describe the problem_

C++ internal segmentation fault when building wxpython
**Version**
master:8a58affe242eb77fb3872cc77a701c6b726332f6

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
N/A

**Logs**
[environment_install_error.txt](https://github.com/ArduPilot/ardupilot/files/15444491/environment_install_error.txt)

Blake-West, 2024-05-25:
Any suggestions mentioned. I'll be happy to look into / work on.

khancyr, 2024-05-25:
Thanks for reporting this. It looks like it compile fine on CI, so we will need to dig out what is the diff

peterbarker, 2024-07-10:
@Blake-West try forcing a different compiler type; unless you want to dip your toes into compiler development, not much else is going to fix an internal compiler error!

Setting `CC` and `CXX` in the environment should be sufficient to try a different compiler.

peterbarker, 2024-07-10:
@Blake-West I just did a `vagrant up noble` here, and it brings up the environment OK, including the compilation of wx

```
(venv-ardupilot) vagrant@vagrant:/vagrant/ArduCopter(master)$ dpkg -l | grep g++ii  g++                                   4:13.2.0-7ubuntu1                       amd64        GNU C++ compiler
ii  g++-13                                13.2.0-23ubuntu4                        amd64        GNU C++ compiler
ii  g++-13-arm-linux-gnueabihf            13.2.0-23ubuntu4cross1                  amd64        GNU C++ compiler for arm-linux-gnueabihf architecture
ii  g++-13-x86-64-linux-gnu               13.2.0-23ubuntu4                        amd64        GNU C++ compiler for x86_64-linux-gnu architecture
ii  g++-arm-linux-gnueabihf               4:13.2.0-7ubuntu1                       amd64        GNU C++ compiler for the armhf architecture
ii  g++-x86-64-linux-gnu                  4:13.2.0-7ubuntu1                       amd64        GNU C++ compiler for the amd64 architecture
```

peterbarker, 2024-07-31:
@Blake-West please also try an `apt update && apt dist-upgrade`

@Blake-West ... or have you already solved this?

Blake-West, 2024-07-31:
Apologies. I did a clean install of ubuntu24 again.
Followed the typical installation instructions and no errors arose as I've typically expected.

Not sure what could have been the issue; maybe new intel CPUs being dodgy?

Regardless, everything seems to be in working order.

khancyr, 2024-07-31:
could happens with a bad packet or anything. thanks to confirm that everything is ok now. 
We will close this.

closed 2024-07-31 by khancyr