ArduPilot/ardupilot #27187: Copter build does not complete Copter-4.5.3

Blake-West, 2024-05-29:
Copter build do not `complete` after configuration.

Ubuntu 22.04.4 clean installation.
Hangs at copying fixed headers for protocol2.0

If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

_Please describe the problem_

**Version**
commit ee14ff3822161368ec403700bf96d02b84def06c
Copter-4.5.3

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadcopter

**Hardware type**
SITL

**Logs**
[log1.txt](https://github.com/ArduPilot/ardupilot/files/15477108/log1.txt)

peterbarker, 2024-07-10:
I've just created a fresh install of 22.04 to test this.

I noticed you configured twice with different compilers.  I suggest `rm -rf build` and *just* try the g++-11 variant of what you were trying to do.

FWIW, not able to replicate here - configure and building with `g++-11` working just fine

peterbarker, 2024-07-31:
Ping @Blake-West did you get this resolved?

Blake-West, 2024-07-31:
Apologies. 
I tried this again with a clean install on ubuntu24 and the error appeared to disappear. For whatever reason, my newest computer was having issues. Not typically had issues installing  my ardupilot.

Appears to be stable.

closed 2024-07-31 by rmackay9