ArduPilot/ardupilot #27193: EK3 Source Switching wrong frame for GUIDED commands

Photronix, 2024-05-29:
Parameters configured to enable EK3 Source switching between SRC1 and SRC3.  SRC1 is Ublox GPS and SRC3 is Mavlink VISION_POSITION_ESTIMATES from visual odometry source.  Switching from SRC1 to SRC3 and back to SRC1 seems to work well.  However, after switching back to SRC1 and sending SET_POSITION_TARGET_GLOBAL_INT (https://mavlink.io/en/messages/common.html#SET_POSITION_TARGET_GLOBAL_INT) the lat/lon values are used as X/Y positions in cm.  Values of GUIP.pX and GUIP.pY show a value in the log equal to the int value of lat/lon sent (x10000000) which is the value typically sent using SRC1.

For some reason the switch from SRC1->SRC3->SRC1 causes a problem.  We have been using this mavlink command for years with no problem using SRC1 only.  In this instance, it resulted in a crash.

[image]

**Version**
Arducopter 4.5.1

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Orange Cube

**Logs**

[L1215_T135_F159_PIX-anon.txt](https://github.com/ArduPilot/ardupilot/files/15487334/L1215_T135_F159_PIX-anon.txt)

Photronix, 2024-05-29:
I should add that we are using MAV_CMD_SET_EKF_SOURCE_SET (42007) to switch EK3 sources (https://mavlink.io/en/messages/ardupilotmega.html#MAV_CMD_SET_EKF_SOURCE_SET)

Photronix, 2024-06-05:
I have completed another test using SITL Arducopter 4.5.3 with (I believe) the same result.
[image]

Photronix, 2024-06-06:
I believe I have resolved the problem. Arducoptet did not switch EKF sources. I believe this problem is on my end.

rmackay9, 2024-06-07:
Hi @Photronix,

Sorry for not replying earlier (I've been off on vacation).  In any case, it's great to hear that you got it sorted.

closed 2024-06-07 by rmackay9