ArduPilot/ardupilot #27196: size of defaults.parm at build time make parameters resets to default at reboot

richaravoil, 2024-05-30:
## Bug report

**Issue details**

The size of the defaults.parm file that you add in your build can make the board reset to default values at each reboot. At first my file was 14KB and contained ~800 parameters. Parameters were resetting to default at each reboot despite me changing them in mission planner. I reduced the size of the defaults to 2KB (~110 parameters) and now the parameters do not reset anymore.
**Version**
Latest version was used

**Platform**
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quadcopter X

**Hardware type**
tested on 2 differents boards : 
+ JHEM_JHEF405 -> reset to defaults if defaults.parm is too big
+ KakuteF7 -> didn't manage to reproduce the bug

amilcarlucas, 2024-05-30:
Usually you do not need to change more than 400 parameters from default.

Are you sure you need to change 800 parameters?

richaravoil, 2024-05-30:
i didn't need to change 800 parameters, it's just that the file was a complete dump of the drone and most of it were default values, that's why i managed to reduce it to 110 parameters

peterbarker, 2024-07-10:
@richaravoil you didn't mention which version of ArduPilot you were testing.  master?  4.5.x?

When you say the board was resetting to default values at each boot - do you really mean, "wasn't using the defaults file"?

richaravoil, 2024-07-10:
> @richaravoil you didn't mention which version of ArduPilot you were testing. master? 4.5.x?
> 
> When you say the board was resetting to default values at each boot - do you really mean, "wasn't using the defaults file"?

Sorry forgot the version, i was testing master. 

i mean that i builded the firmware with a default.parm file to override ardupilot defaults with mine. i flashed the board, was working fine. I realised some of my parameters needed tuning so i modified them in MP and wrote them to the board. I was modifying the min/max of one of my servos so i could see it taken in consideration by the board. After a reboot, the min/max values i just modified are back to the values defined in the file i used at build time 

Hope it helps with the full explanation