ArduPilot/ardupilot #27197: Custom firmware for RealFlight and Mission planner doesn't work as per instructions

Georacer, 2024-05-30:
## Bug report

**Issue details**

The SITL binary for Windows that is compiled by Cygwin does not work, as per the instructions in:
https://ardupilot.org/dev/docs/sitl-with-realflight.html#using-your-code-branch-with-mission-planner-sitl-and-realflight

When Mission Planner launches the binary for simulation it reports "Failed to connect to SITL instance"
[image]

Compiling with a local Cygwin toolchain also fails with the same mode.

Relevant discussions have been reported in:
- https://discord.com/channels/674039678562861068/710994174299603055/1245286277323165738
- https://discord.com/channels/674039678562861068/797952826847920128/1245405919383388373

The workflow seems to work with the current stable release 4.5.2.

**Version**
Current master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Airplane, Tailsitter

**Hardware type**
N/A

**Logs**
N/A

EShamaev, 2024-07-01:
SITL for MP got broken completely.
[image]
When selecting "Stable" all works.
When selecting "Latest" the sim opens for one seconds and closes immidiatelly without anything in messages.

EShamaev, 2024-07-01:
Narrowed down to exact cause by wrapper over malloc.

I am not too good at python and don't have expertise with waf system so for a quick solution following was made to solve the issue:

```
diff --git a/Tools/ardupilotwaf/boards.py b/Tools/ardupilotwaf/boards.py
index 5f5ce5434b..4c763aee0e 100644
--- a/Tools/ardupilotwaf/boards.py
+++ b/Tools/ardupilotwaf/boards.py
@@ -754,7 +754,7 @@ class sitl(Board):
 
         # wrap malloc to ensure memory is zeroed
         # don't do this on MacOS as ld doesn't support --wrap
-        if platform.system() != 'Darwin':
+        if platform.system() != 'Darwin' and cfg.env.DEST_OS != 'cygwin':
             env.LINKFLAGS += ['-Wl,--wrap,malloc']
         
         if cfg.options.enable_sfml:
diff --git a/libraries/AP_Common/c++.cpp b/libraries/AP_Common/c++.cpp
index 8ab798744f..4b93f26243 100644
--- a/libraries/AP_Common/c++.cpp
+++ b/libraries/AP_Common/c++.cpp
@@ -85,7 +85,7 @@ void operator delete[](void * ptr)
 }
 
 
-#if CONFIG_HAL_BOARD != HAL_BOARD_CHIBIOS
+#if CONFIG_HAL_BOARD != HAL_BOARD_CHIBIOS && CONFIG_HAL_BOARD != HAL_BOARD_SITL
 /*
   wrapper around malloc to ensure all memory is initialised as zero
   ChibiOS has its own wrapper
```

I hope @tridge will find a more elegant solution that will deserve a PR.

PR #27473 (2024-07-06): build: fixed cygwin build with malloc wrap
on Cygwin we need to wrap _malloc_r not malloc
See https://cygwin.com/pipermail/cygwin/2000-July/038916.html
fixes #27197
closed 2024-07-06 by tridge