ArduPilot/ardupilot #27205: Rover: Add RTL_ENABLE parameter

yuri-rage, 2024-05-31:
## Feature request

Request an RTL_ENABLE parameter for Rover that will allow users to opt out of all RTL behaviors by setting this parameter to zero.

**Is your feature request related to a problem? Please describe.**

Large, land-based vehicles rarely if ever have a valid use case for RTL.

Recently, a user had a large mower accidentally activate RTL mode, which could have been catastrophic. Making a beeline for home could have dire effects if RTL is accidentally activated (either by failsafe, mission command, or errant mode change). Of course, this is all user error related, but if a vehicle should never be able to enter RTL, there should be a way to signal that to the autopilot and prevent such error from ever occurring.

**Describe the solution you'd like**

Parameter as described in opening statement.

**Describe alternatives you've considered**

I have written a Lua script to detect errant RTL activation and immediately switch to HOLD mode. However, this is only valid for scripting enabled autopilots, so it is not a universal solution.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[x ] Rover
[  ] Submarine

**Additional context**

Relevant Discord discussion with @rmackay9:
https://discord.com/channels/674039678562861068/797946854792757259/1245931364134551552

Relevant Mission Planner issue discussing the method by which the user accidentally stumbled into RTL:
https://github.com/ArduPilot/MissionPlanner/issues/3358

IamPete1, 2024-05-31:
We do have `FLTMODE_GCSBLOCK`, it does have a RTL bit. That would prevent switching to RTL from the GCS. Of course there are still failsafes...

yuri-rage, 2024-05-31:
> We do have `FLTMODE_GCSBLOCK`, it does have a RTL bit. That would prevent switching to RTL from the GCS. Of course there are still failsafes...

Would not prevent an RTL from auto, which was the root cause here. Other suggestions include proper use of inclusion/exclusion fences. I understand we can't save everyone from themselves, but an opt-in feature like this might well be of value.

muramura, 2024-05-31:
Without preparing a new flag, I think it can be set to each failsafe action.

https://ardupilot.org/rover/docs/rover-failsafes.html#failsafes

yuri-rage, 2024-05-31:
> Without preparing a new flag, I think it can be set to each failsafe action.
> 
> I think SMART RTL is better than RTL. SMART RTL determines the shortest return path based on the travel history. It doesn't return in a straight line like RTL.
> 
> https://ardupilot.org/rover/docs/rover-failsafes.html#failsafes

Correct on both counts, however...

Failsafe actions will not preclude accidental entry to RTL via an errantly applied RTL waypoint during an auto mission (which is really the intent of the feature request). And SRTL still has many pitfalls and a lack of application on the vehicles in question. For example, but not limited to: SRTL memory fills during prolonged missions and subsequently fails.

geofrancis, 2024-05-31:
I drive my rovers around town, if it ever tried to RTL it would either end up on the road or into a wall so I would enable this if it was an option.

IamPete1, 2024-05-31:
I don't think RTL is the default action for any of the failsafe', they either report only or go into hold. We can protect against accidental switch from the GCS with `FLTMODE_GCSBLOCK`.

So the user has to setup a RC switch or configured a failsafe action to RTL.

Preventing entering RTL from a mission command could not be enabled by default, because that would break existing setups. So it would only save people after the first crash when they go and read about how they could have prevent it.

A lua script to add a pre-arm for RTL points in the mission would be quite straight forward. (or the very simple script suggested in the issue)

Fence and avoidance sensors might prevent the crash if not the RTL itself.

yuri-rage, 2024-06-01:
> Preventing entering RTL from a mission command could not be enabled by default

Agree. My suggestion is an RTL_ENABLE parameter with 1 as the default value, such that it would not be a breaking change.