ArduPilot/ardupilot #27208: SERVO-BLH-3DMASK and SERVO-BLH-RVMASK encountered conflicts

burgeruser, 2024-05-31:
## Bug report
I have experience with BLheli-s and BLheli32 [ESCs](https://ardupilot.org/copter/docs/common-dshot-escs.html). When setting both SERVO-BLH-3DMASK and SERVO-BLH-RVMASK, I encountered conflicts. If a flight controller output channel is connected to an ESC and both masks are set, the SERVO-BLH-RVMASK appears to be ineffective. This greatly complicates my assembly process, as it may require disassembling and rewiring the ESCs and motors if they cannot be conveniently flashed with BLheli firmware via its [configuration interface](https://ardupilot.org/copter/docs/common-blheli32-passthru.html).
[image]

The ArduPilot wiki previously mentioned a turtle mode for multicopters, stating that bidirectional ESCs could enable flips. However, following the instructions led to issues, as there were numerous bugs.

**Version**
I have indeed tested this on Copter-4.4 and Copter-4.5.0 platforms, but I haven't tested it on other versions, and it might apply to them as well, which would require actual testing. Additionally, I recommend testing the [turtle mode](https://ardupilot.org/copter/docs/turtle-mode.html) since it may exhibit similar behavior in other firmware versions.

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X4

**Hardware type**
Pixhawk、CubeOrange

Hwurzburg, 2024-05-31:
A reversed esc and an esc operating in reversible mode are mutually exclusive features.....just as they are when using the BLHeli configurator

burgeruser, 2024-06-01:
BLHeli configurator can set-up ESC both bi-directional mode, and also configurate reversible mode. 
But if I using a new ESC without being set-up,it is not be able to runing  right, when setting both SERVO-BLH-3DMASK and SERVO-BLH-RVMASK in Missionplanner.

burgeruser, 2024-06-01:
> A **reversed** unidirectional mode esc and an esc operating in **reversible** mode are mutually exclusive features.....just as they are when using the BLHeli configurator

BLHeli configurator can set-up ESC both bi-directional mode, and also configurate reversible mode.
But if I using a new ESC without being set-up in BLHeli configurator,it is not be able to runing right, when setting both SERVO-BLH-3DMASK and SERVO-BLH-RVMASK in MissionPlanner.

Hwurzburg, 2024-06-01:
your terms are confusing...bi-directional mode has nothing to do with unidirectional reversing or reversible operation ...bi-directional mode in BLHeli32 is telemetry mode...