ArduPilot/ardupilot #27227: SITL build fails on alpine for Copter 4.5

Zavitit, 2024-06-03:
## Bug report

**Issue details**

Hi i’m trying to build SITL using Copter 4.5 and for some reason the compilation fails with this error:

```
./../libraries/AP_GPS/AP_GPS_SITL.cpp:37:27: error: aggregate 'timeval first_tv' has incomplete type and cannot be defined
    37 |     static struct timeval first_tv;
       |                           ^~~~~~~~
 compilation terminated due to -Wfatal-errors.

 Waf: Leaving directory `/ardupilot/src/build/sitl'
 Build failed
  -> task in 'objs/AP_GPS' failed (exit status 1): 
  {task 135805090486368: cxx AP_GPS_SITL.cpp -> AP_GPS_SITL.cpp.0.o}
: 
```
I am using alpine 3.19 with a configuration that used to work with SITL from Copter 4.4.3 (altho with alpine 3.16 but the error is the same in both versions)

this issue is opened in response to: https://discuss.ardupilot.org/t/sitl-build-fails-on-alpine/118072
however the issue details are duplicated here for clarity and documentation 

**Version**
Copter 4.5

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

peterbarker, 2024-07-10:
Thanks for the report.

I've created a PR to resolve the issue - feel free to test the PR, but I do intend to merge it on CI pass.

closed 2024-07-10 by peterbarker

Zavitit, 2024-07-22:
@peterbarker Thank you very much for the help but unfortunately  I'm still having this issue 

I'm adding my Dockerfile maybe I'm just doing something wrong?
```dockerfile
FROM alpine:3.19 AS builder

WORKDIR /ardupilot

RUN apk update && apk add --no-cache \
        linux-headers \
        g++ \
        python3 \
        py-future \
        py-pip \
        libxml2-dev \
        libxslt-dev \
        git \
    && ln -sf python3 /usr/bin/python \
    &&  rm -rf /var/cache/apk/*

RUN python3 -m pip install --user --no-deps --no-cache-dir empy==3.3.4 pexpect ptyprocess --break-system-packages

RUN git clone https://github.com/ardupilot/ardupilot.git --depth 1 --single-branch src \
    && cd src \
    && git submodule update --init --recursive --depth 1

RUN cd /ardupilot/src && ./waf configure --board sitl --no-submodule-update \
    && ./waf copter
# fails during ./waf copter

ENTRYPOINT ["sh"]
```
i did some manual experimentation and just rerunning the `./waf copter` command resusted in around 5 targets that won't compile
output for the last run:
```
Waf: Entering directory `/ardupilot/src/build/sitl'
Embedding file locations.txt:Tools/autotest/locations.txt
Embedding file models/Callisto.json:Tools/autotest/models/Callisto.json
Embedding file models/freestyle.json:Tools/autotest/models/freestyle.json
Embedding file models/plane-3d.parm:Tools/autotest/models/plane-3d.parm
Embedding file models/plane.parm:Tools/autotest/models/plane.parm
Embedding file models/xplane_heli.json:Tools/autotest/models/xplane_heli.json
Embedding file models/xplane_plane.json:Tools/autotest/models/xplane_plane.json
Generating compile_commands.json
Build commands will be stored in build/sitl/compile_commands.json
Generating compile_commands.json
Build commands will be stored in build/sitl/compile_commands.json
[ 802/1331] Compiling libraries/SITL/SIM_VectorNav.cpp
[ 843/1331] Compiling libraries/SITL/SIM_MicroStrain7.cpp
[ 844/1331] Compiling libraries/SITL/SIM_MicroStrain.cpp
[ 873/1331] Compiling libraries/SITL/SIM_GPS_NMEA.cpp
[ 884/1331] Compiling libraries/SITL/SIM_GPS_MSP.cpp
[ 889/1331] Compiling libraries/SITL/SIM_GPS.cpp
[ 893/1331] Compiling libraries/SITL/SIM_FlightAxis.cpp
[1073/1331] Compiling libraries/AP_HAL_SITL/SITL_cmdline.cpp
../../libraries/SITL/SIM_GPS_NMEA.cpp: In member function 'virtual void SITL::GPS_NMEA::publish(const SITL::GPS_Data*)':
../../libraries/SITL/SIM_GPS_NMEA.cpp:39:20: error: aggregate 'SITL::timeval tv' has incomplete type and cannot be defined
   39 |     struct timeval tv;
      |                    ^~
compilation terminated due to -Wfatal-errors.

../../libraries/SITL/SIM_MicroStrain.cpp: In static member function 'static void SITL::MicroStrain::simulation_timeval(SITL::timeval*)':
../../libraries/SITL/SIM_MicroStrain.cpp:44:27: error: aggregate 'SITL::timeval first_tv' has incomplete type and cannot be defined
   44 |     static struct timeval first_tv;
      |                           ^~~~~~~~
compilation terminated due to -Wfatal-errors.

../../libraries/SITL/SIM_GPS_MSP.cpp: In member function 'virtual void SITL::GPS_MSP::publish(const SITL::GPS_Data*)':
../../libraries/SITL/SIM_GPS_MSP.cpp:57:20: error: aggregate 'SITL::timeval tv' has incomplete type and cannot be defined
   57 |     struct timeval tv;
[... 35 lines cut ...]
 -> task in 'objs/SITL' failed (exit status 1): 
        {task 137329632954304: cxx SIM_GPS_NMEA.cpp -> SIM_GPS_NMEA.cpp.0.o}
 (run with -v to display more information)
 -> task in 'objs/SITL' failed (exit status 1): 
        {task 137329632958112: cxx SIM_MicroStrain.cpp -> SIM_MicroStrain.cpp.0.o}
 (run with -v to display more information)
 -> task in 'objs/SITL' failed (exit status 1): 
        {task 137329632954192: cxx SIM_GPS_MSP.cpp -> SIM_GPS_MSP.cpp.0.o}
 (run with -v to display more information)
 -> task in 'objs/SITL' failed (exit status 1): 
        {task 137329632953968: cxx SIM_GPS.cpp -> SIM_GPS.cpp.0.o}
 (run with -v to display more information)
 -> task in 'objs/SITL' failed (exit status 1): 
        {task 137329632953072: cxx SIM_FlightAxis.cpp -> SIM_FlightAxis.cpp.0.o}
 (run with -v to display more information)
 -> task in 'objs/SITL' failed (exit status 1): 
        {task 137329632958224: cxx SIM_MicroStrain7.cpp -> SIM_MicroStrain7.cpp.0.o}
 (run with -v to display more information)
 -> task in 'objs/SITL' failed (exit status 1): 
        {task 137329632964496: cxx SIM_VectorNav.cpp -> SIM_VectorNav.cpp.0.o}
 (run with -v to display more information)
 -> task in 'objs/AP_HAL_SITL/ArduCopter' failed (exit status 1): 
        {task 137329632802448: cxx SITL_cmdline.cpp -> SITL_cmdline.cpp.3.o}
 (run with -v to display more information)
```