ArduPilot/ardupilot #27263: Unreliable mode switch via aux function

LupusTheCanine, 2024-06-06:
## Bug report

**Issue details**

Rapidly switching between two aux function modes causes mode to stay in mode selected by FLTMODE_CH
Flight modes are assigned to channels 8 & 9
FBWA on 8
RTL on 9
FLTMODE_CH is 5

Going from RTL to FBWA is typically fine
Going from FBWA to RTL transients through RTL ends in whatever is selected on FLTMODE_CH.

**Version**
Arduplane 4.5.3

**Platform**
[?] All
[?] AntennaTracker
[?] Copter
[*] Plane
[?] Rover
[?] Submarine

**Airframe type**
Doesn't apply

**Hardware type**
Matek H743 Wing 
Radiomaster TX16S
FrSky XJT with X8R receiver SBUS rc + S.Port Telemetry

**Logs**
[image]
Red is FBWA
Green is RTL
[97 01-01-1970 01-00-00.zip](https://github.com/user-attachments/files/15687674/97.01-01-1970.01-00-00.zip)

rmackay9, 2024-06-07:
Hi @LupusTheCanine,

In general it should be "edge detected" and the last change wins.  For the aux mode functions that implement a flight mode, when they are moved to the low position they will switch back to whatever the flight mode channel is set to.

There is also some "debouncing" in the code to ensure that very short transient RC inputs from the flight mode channel and aux function are ignored.

TBH, I suspect that there's no problem in the code.. but perhaps I'm misunderstanding the report..

LupusTheCanine, 2024-06-07:
@rmackay9 

The problem is that simultaneous transition doesn't work reliably. 
I can go RTL->FBWA just fine without even entering Manual but going the other way causes 
the transition to be FBWA->RTL->Manual.

Rising edge happening within the debounce time should cancel falling edge mode reversion,

I would classify this as unexpected behaviour with potentially dangerous consequences.

I want to have quick access "Emergency" modes on a separate switch as while buttons on the TX16S are great for switching modes when you aren't in the hurry, they don't really cut it in emergency.

Having one three position "override" switch would be nice alternative but I would rather prefer the sequencing fixed.

IamPete1, 2024-06-07:
We check the switches in order, so FBWA on 8 is checked before RTL on 9. If you switch them round I would expect your results to be the other way too. As far as I can tell ArduPilot is handling it consistently. Don't forget RC is not the only source of mode changes.

> I want to have quick access "Emergency" modes on a separate switch as while buttons on the TX16S are great for switching modes when you aren't in the hurry, they don't really cut it in emergency.

Presumably your not planning on having to switch two switches in this case then? The issue is because you have done a transmitter mix of some sort? Why not fix it in your mix?

LupusTheCanine, 2024-06-07:
> Presumably your not planning on having to switch two switches in this case then? The issue is because you have done a transmitter mix of some sort? Why not fix it in your mix?

I would not consider adding a delay to mode change switch an acceptable solution especially given that frame loss can potentially wreck the timing anyways. 
"Mode enable" switch going high should override any pending mode enable going down preferably while retaining the original previous mode.

LupusTheCanine, 2024-07-05:
> Hi @LupusTheCanine,
> 
> In general it should be "edge detected" and the last change wins. For the aux mode functions that implement a flight mode, when they are moved to the low position they will switch back to whatever the flight mode channel is set to.

I did some further testing while implementing a TX side patch and about 0.4s delay is required for mode switch to happen correctly under ideal conditions.

IMHO mode restore shouldn't happen if there is another mode switch activating at the same time.

IamPete1, 2024-07-06:
@LupusTheCanine as you can probably tell, were not fantastically keen to change the behaviour here. 

Can you explain what your trying to do, maybe we can come up with a alternate solution. Why not use the 6 position mode switch? We have had longstanding requests to allow more than 6 modes on the switch, would that be a way forward for you?

LupusTheCanine, 2024-07-06:
Yes, I am trying to set up quick access mode switch on my TX16S. It has a very nice 6 pos mode switch that works great for switching between assisted/automatic modes but is quite cumbersome to use in an emergency situation as it requires taking a hand off either stick or throttle. 
I want to setup an additional three-position switch that will allow me to access two flight modes quickly (ex. MANUAL and RTL). The requirement is that the switch happens reliably even when going from one quick access mode to the other. Unfortunately, current implementation of mode-select aux function tends to execute restore previous mode after switch to the new mode.

Ability to configure more modes on FLFMODE_CH would definitely solve the issue as I could configure non timer dependant mix for mode selection. I would suggest setting maximum limit to 16 (Express LRS channel 12 resolution in hybrid mode).

IMHO reverting being the dominant behaviour is a bug and I wouldn't expect simultaneous selection of mode A and deselection of mode B to take me anywhere but to mode A.

IamPete1, 2024-07-06:
> I want to setup an additional three-position switch that will allow me to access two flight modes quickly (ex. MANUAL and RTL). The requirement is that the switch happens reliably even when going from one quick access mode to the other.

I guess you don't have manual and RTL on your 6 position switch? So you can't just do a mixer there?

The reason we haven't rushed to implement more than 6 so far is that 6 modes is already quite a lot for most users to keep track of.

> reverting being the dominant behaviour

It is not the dominant behavior, it depends on the channel number, if you switch both at once then the switch on the highest channel wins because we check them in numerical order.

BTW, you mentioned your running a H7, custom mode switches that behave however you like would be a simple lua script.