ArduPilot/ardupilot #27268: Relay switch for video transmitter voltage control not working in 4.5 (RADIX2HD)

mluessi, 2024-06-07:
The RADIX2HD uses the Relay2 functionality to power on/off the voltage regulator for the video transmitter. This was working correctly in 4.4 but no longer works as expected in 4.5.  By default, when no RC channel is mapped to the relay, the video transmitter should be powered on, but it is powered off.  When mapping an RC channel to the relay (e.g. `RC8_OPTION = 34`), the relay can only be toggled once.

The default state of the relay can be fixed by using `RELAY2_DEFAULT = 1`. The target currently has `RELAY_DEFAULT = 1` in its `defaults.param` file and I assume this needs to be changed there. With that changed, the video transmitter voltage turns on when no RC channel is mapped to the relay. The relay can also be controlled from MissionPlanner.

However, controlling the relay using an RC channel still does not work. When using channel 8 ( `RC8_OPTION = 34`) the relay can only toggled once (on to off) using the switch on the RC transmitter. I confirmed that the GPIO pin does not toggle, and this has been confirmed with several flight controllers. So it does not seem to be a hardware problem but a bug or misconfiguration.

The relevant parameters are:

    RELAY2_PIN = 81
    RELAY2_FUNCTION = 1
    RELAY2_DEFAULT = 1

For control with RC channel 8:

    RC8_OPTION = 34

**Version**
not working in 4.5.3, 4.5.2, 4.5.1, working in 4.4.4

**Platform**
[  ] All
[  ] AntennaTracker
[ x] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
NA

**Hardware type**
[RADIX2HD](https://ardupilot.org/copter/docs/common-radix2hd.html)

**Logs**
NA

IamPete1, 2024-06-07:
Can you upload a log of the test please. You can enable disarm logging rather than arming.

mluessi, 2024-06-08:
@IamPete1 I recorded a log. It booted with RC8 high and then I toggled it a few times every 10 s. I can see RCIN.C8 changing, but not sure what I need to look at to see if the relay should be toggling. 

Log file: [radix2hd_vreg_switch_test_20240608.log](https://drive.google.com/file/d/1F1UZU7_VJyz_WIgOw0S5gvDY4oTiLG2v/view?usp=sharing)

IamPete1, 2024-06-08:
Your switch is conflicting with the flight mode. ` FLTMODE_CH` is 8.

mluessi, 2024-06-08:
@IamPete1 well, that was silly of me. It works when using channel 7.

But, can you confirm that `RELAY_DEFAULT` needs to be changed to `RELAY2_DEFAULT` in the default parameters for this target? There also seem to be several other targets that may need to be changed. See below

```
libraries/AP_HAL_ChibiOS/hwdef/KakuteH7v2/defaults.parm:RELAY_DEFAULT 1
libraries/AP_HAL_ChibiOS/hwdef/skyviper-v2450/defaults.parm:RELAY_DEFAULT 1
libraries/AP_HAL_ChibiOS/hwdef/MambaH743v4/defaults.parm:RELAY_DEFAULT 1
libraries/AP_HAL_ChibiOS/hwdef/SkystarsH7HD-bdshot/defaults.parm:RELAY_DEFAULT 1
libraries/AP_HAL_ChibiOS/hwdef/skyviper-journey/defaults.parm:RELAY_DEFAULT 1
libraries/AP_HAL_ChibiOS/hwdef/KakuteH7Mini/defaults.parm:RELAY_DEFAULT 1
libraries/AP_HAL_ChibiOS/hwdef/MambaF405-2022/defaults.parm:RELAY_DEFAULT 1
libraries/AP_HAL_ChibiOS/hwdef/RADIX2HD/defaults.parm:RELAY_DEFAULT 1

```

IamPete1, 2024-06-12:
> But, can you confirm that `RELAY_DEFAULT` needs to be changed to `RELAY2_DEFAULT` in the default parameters for this target? There also seem to be several other targets that may need to be changed. See below

Yes, you are correct, I have opened a PR that fixes. https://github.com/ArduPilot/ardupilot/pull/27286

mluessi, 2024-06-17:
@IamPete1 thanks!

closed 2024-06-17 by rmackay9