ArduPilot/ardupilot #27308: Feat: Request for controlling the PWM values of all motors for autonoumous tracking

sairaghavamunagala2001, 2024-06-15:
# Feature Request: Synchronized PWM Control for All Motors
## Platform
- Copter
## Description:
The requested feature aims to allow simultaneous control of the PWM (Pulse Width Modulation) values for all motors on a drone. This capability is crucial for enabling autonomous tracking, which is a fundamental component of reinforcement learning applications in drone technology.

## Problem Statement:
Currently, it may not be possible to control all motor PWM values in a synchronized manner through the existing interfaces. This limitation hinders the development of advanced autonomous tracking algorithms, which require precise and coordinated motor control.

## Proposed Solution:
Introduce a new interface or extend the existing one to allow batch updates of PWM values for all motors. This interface should:

- Accept an array of PWM values corresponding to each motor.
- Apply the PWM values simultaneously to ensure synchronized motor behavior.
- Provide real-time feedback on the applied PWM values to facilitate debugging and tuning.
## Use Case: Reinforcement Learning for Autonomous Tracking
In reinforcement learning, an agent (in this case, the drone) learns to perform tasks through trial and error, receiving rewards or penalties based on its actions. Precise and synchronized control of motor speeds is essential for:

- Stabilizing the drone during complex maneuvers.
- Executing coordinated movements required by the learning algorithms.
- Gathering consistent data for training the reinforcement learning models.
### API Design:

- Define a clear and concise API for setting PWM values for all motors.
- Ensure the API can accept an array of PWM values corresponding to each motor.
- Provide mechanisms to handle exceptions and errors gracefully.
- Integrate with existing control systems to ensure compatibility and ease of use.
### Latency and Synchronization:

- Ensure minimal latency and perfect synchronization when applying PWM values to all motors.
- Implement efficient data transmission mechanisms to reduce communication delays.
- Use timestamping or other synchronization techniques to ensure that all motors receive their PWM signals simultaneously.
### Safety Checks:

- Implement safety checks to prevent out-of-range PWM values that could damage the motors or compromise flight stability.
- Include validation routines to verify that PWM values are within acceptable limits before applying them.
- Provide feedback and logging mechanisms to monitor PWM values and detect any anomalies in real-time.
### pymavlink Support:

- Utilize pymavlink to send MAVLink commands for setting PWM values for all motors.
- Ensure the pymavlink implementation can handle the simultaneous control of multiple motors.
- Maintain compatibility with various MAVLink-capable flight controllers.
- Provide examples and documentation for users to easily integrate pymavlink into their projects.
- Implement robust error handling and reconnection logic to ensure reliable communication with the drone.

khancyr, 2024-06-15:
Don't trust AI and learn the code by yourself. We can already do this

closed 2024-06-15 by khancyr

lthall, 2024-06-15:
If you are manually controlling all PWM outputs then why do you need copter at all?

I agree with the closing of this feature request.

sairaghavamunagala2001, 2024-06-15:
I need support from the pymavlink or from API side, so onboarding processor will control all PWM values @lthall @khancyr with respect to the detected object.

rmackay9, 2024-06-15:
I think what @sairaghavamunagala2001 is trying to do is train an AI to fly a multicopter directly without going through AP's estimators and controllers.  I've heard this idea from a Japan based AI researcher as well a few years ago so I think it is a reasonable thing to try and support.

I think much like someone extending AP's estimators or controllers they don't want to take over all control immediately and they also want to be sure to have a fall back to regular operation when things go wrong.

Regardless of whether AI is used or not, I think the request is for a PWM pass-through from an external companion computer.  We already [support this in Lua](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/generator/description/bindings.desc#L364).

@sairaghavamunagala2001 you've seen our [developer focused MAVLink interface wiki page](https://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html)?  Also feel free to ping me in AP's discord perhaps within the "ai" channel.  Make sure to put "@rmackay9" in your post so I don't miss it.

sairaghavamunagala2001, 2024-06-16:
okay sure, thanks for the input @rmackay9 .