ArduPilot/ardupilot #27314: Problem with micro ros. (won't publish the topic).

AlexanderBVKtex, 2024-06-16:
## Bug report

**Issue details**
When following the instructions ([ROS2 with sitl](https://ardupilot.org/dev/docs/ros2-sitl.html#ros-2-with-sitl)), micro ros does not publish the topic. This is due to the error "micro ros agent" appearing in the terminal.

**Version**
Ardupilot 4.5.4; Some previous versions do not have this problem

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**System**
Ubuntu 22.04 arm64
SITL with ROS2

**Logs**
[logs.txt](https://github.com/user-attachments/files/15859880/logs.txt)

rmackay9, 2024-06-16:
FYI @Ryanf55 @srmainwaring

rmackay9, 2024-06-16:
@AlexanderBVKtex,

Thanks for the report.  In general it is better to post in the support forums first and then after they've been confirmed as real bugs or enhancement requests we can add them here.

Ryanf55, 2024-06-17:
The change was not backported to 4.5 to remove the refs file. I'm on vacation for the next two weeks, but in the mean time, please try removing that argument in your command.

Ryanf55, 2024-06-17:
See https://github.com/ArduPilot/ardupilot/pull/27145

rmackay9, 2024-06-17:
Great, I will go ahead and close this although we will still see replied. txs!

closed 2024-06-17 by rmackay9