ArduPilot/ardupilot #27315: Upgrade to AP 4.5.x causes external I2C baro BMP280 not detected while onboard SPI BMP280 works.

RBirdie001, 2024-06-16:
## Bug report

**Issue details**

Plane with FC Omnibus F4 upgraded from Arduplane 4.4.4 to 4.5.3.
Beside onboard SPI BMP280 baro I have external I2C baro also BMP280.
With previous versions including the 4.4.4 external baro worked but after upgrading to 4.5.2 (4.5.3) it’s not detected anymore.
My settings: BARO_EXT_BUS = -1 and BARO_PROBE_EXT = 2.
For the test I reverted back to 4.4.4 without touching the configuration and ext. baro started working again.
Both baros are the same type so hardware driver should be there already. 
Tried to discuss here but solution not found:
https://discuss.ardupilot.org/t/external-baro-bmp280-stopped-working-after-upgrading-from-ap4-4-4-to-4-5-3/119075 

**Version**
Arduplane  4.5.2 and 4.5.3

[**Platform**]
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
V-tail plane - Mini Talon 

**Hardware type**
Omnibus F4V3

**Logs**
_Please provide a link to any relevant logs that show the issue_

peterbarker, 2024-07-10:
This board became minimized due to size constraints (we like the firmware to be able to fit on the boards!), and as part of that probing for external barometers was removed.

It's *probably* a legitimate target for removal; external i2c baros on fpv/osd boards with only 1MB for flash are going to be a rarity, I'm afraid.

I've created this PR to add an option to our custom build server to get the feature back: https://github.com/ArduPilot/ardupilot/pull/27516

In the meantime you can build your own firmware, or I could potentially knock you something up.

peterbarker, 2024-07-10:
... incidentally, thanks for the detailed report.

RBirdie001, 2024-07-11:
Thanks, it would be great to get this feature back! I'm using external baro mainly as an external thermometer.
I had the idea to use custom build server to get the ext. baro support back, but in this case when both baros are the same type it's not possible. I don't have enough knowledge to build a custom firmware completely myself from the scratch.

RBirdie001, 2024-07-12:
I noticed, that after merging pr [ ( https://github.com/ArduPilot/ardupilot/pull/27516)](url) in the custom build server is now available option EXTERNALAHRS Barometric Sensor which I hope should enable searching for external baros. I ticked it (left the rest default) and created a custom build [https://custom.ardupilot.org/builds/plane:omnibusf4:f24dd1d75fc894c032342c9ef4076e9dfa7aff86:efebab605db38bab8e7989ea453129c3/](url)
Uploaded to my plane, but external baro is still not detected.
Do I understand something wrong or it doesn't work?

peterbarker, 2024-11-24:
Sorry for the slow response.

That's not the right option - this is the one you're after:

[image]

... obviously you also need to enable the external baro you want to detect!

Please let us know how you get on...

peterbarker, 2025-05-11:
I'm going to assume @RBirdie001 sorted things out somehow here.

Please re-open if still issues here.

closed 2025-05-11 by peterbarker