ArduPilot/ardupilot #27316: OBAL startup problem. (SIGSEGV, Segmentation fault.)

AlexanderBVKtex, 2024-06-16:
## Bug report

**Issue details**
After running my compiled executable for OBAL, the following error appears: SIGSEGV, Segmentation fault. (Everything is done according to OBAL's [instructions](https://github.com/HefnySco/OBAL/wiki/06---Build-and-Install).)
When debugged, it gives the following output:
```
RPI 4 

Program received signal SIGSEGV, Segmentation fault.
AP::rtc () at ../../libraries/AP_RTC/AP_RTC.cpp:309
309	{
```

**Version**
Ardupilot 4.5.4; No problems on Ardupilot 4.3.3

**Platform**
[?] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine
_**Only checked for copter, probably applies to all types.**_

**System**
Ubuntu 22.04 arm64
Compiled with clang++ 14.0.0

HefnySco, 2024-06-19:
Please confirm that CONFIG_STRICT_DEVMEM is enabled.
as mentioned here https://github.com/HefnySco/OBAL/wiki/06---Build-and-Install.

I made a fast test on RPI-Zero W2 with 2021-05-07-raspios-buster-armhf-lite.zip and every thing seems to be working fine.

HefnySco, 2024-06-19:
int clock_gettime(clockid_t clockid, struct timespec *ts)
{
    //! @todo the value of clockid is ignored here.
    //! A fallback mechanism is employed against the caller's choice of clock.
    uint64_t utc_usec;
    if (!AP::rtc().get_utc_usec(utc_usec)) {
        utc_usec = AP_HAL::micros64();
    }
    ts->tv_sec = utc_usec / 1000000ULL;
    ts->tv_nsec = (utc_usec % 1000000ULL) * 1000UL;
    return 0;
}

peterbarker, 2024-07-10:
@HefnySco that looks like infinite recursion, which is generally considered Bad