ArduPilot/ardupilot #27340: Sensor Position Offset Compensation accel bug

lixiaowei789, 2024-06-20:
Only position and velocity compensation has been done, does acceleration need to be compensated? Otherwise, it can easily cause turbulence

code:
   // If the IMU accelerometer is offset from the body frame origin, then calculate corrections
    // that can be added to the EKF velocity and position outputs so that they represent the velocity
    // and position of the body frame origin.
    // Note the * operator has been overloaded to operate as a dot product
    if (!accelPosOffset.is_zero()) {
        // calculate the average angular rate across the last IMU update
        // note delAngDT is prevented from being zero in readIMUData()
        Vector3F angRate = dal.ins().get_gyro(gyro_index_active).toftype();

        // Calculate the velocity of the body frame origin relative to the IMU in body frame
        // and rotate into earth frame. Note % operator has been overloaded to perform a cross product
        Vector3F velBodyRelIMU = angRate % (- accelPosOffset);
        velOffsetNED = Tbn_temp * velBodyRelIMU;

        // calculate the earth frame position of the body frame origin relative to the IMU
        posOffsetNED = Tbn_temp * (- accelPosOffset);
    } else {
        velOffsetNED.zero();
        posOffsetNED.zero();
    }

lixiaowei789, 2024-06-20:
Recently, I encountered the phenomenon of vertical turbulence when installing the flight control at a distance of 12cm from the center of the rack at the tail end. It was found that there was a sudden change in acceleration, and the main reason was analyzed to be the sudden change in vertical acceleration caused by the action of the lever arm

rmackay9, 2024-06-20:
This has come up in the past with @priseborough and my understanding is that accelerometer compensation based on the position offset cannot be done well enough to be useful.  I'm unsure of the exact reason but it is not a *bug* so I've changed the title.

IamPete1, 2024-06-20:
https://github.com/ArduPilot/ardupilot/pull/23511

The problem is that gyro noise is amplified to make large accelerations and it makes everything worse.

lixiaowei789, 2024-06-21:
> #23511
> 
> The problem is that gyro noise is amplified to make large accelerations and it makes everything worse.

I placed the flight control at the rear of the rack 12cm away from the center, is it related to the gyroscope? My idea is to approximate the corrected acceleration value based on the position change of each cycle, so that it can be optimized