ArduPilot/ardupilot #27357: Plane: v4.6-dev. Cameras does not switching

knoppixmeister, 2024-06-22:
Analog video cameras does not switching on ArduPlane 4.6-dev as described here:
https://ardupilot.org/copter/docs/common-matekf405-te.html#camera-and-supply-switch 

but tested on ArduPlane-4.4.1 and everything works.

Same problem appears also on Matek H743

**Version**
ArduPlane 4.6.0-dev

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Plane

**Hardware type**
Matek F405-TE

IamPete1, 2024-06-22:
On the H743 please try setting `RELAY2_DEFAULT` to 1.

knoppixmeister, 2024-06-22:
Yes. With RELAY2_DEFAULT=1 on Matek H743 everything works fine.

Note: Users of Matek H743 also should choose voltage and solder Vsw bridge

IamPete1, 2024-06-23:
The same thing should work on the Matek F405-TE, but I would have expected that to have come across in the update if you had enabled it on 4.4.