ArduPilot/ardupilot #27372: Issue with H-RTK Mosaic-H (Dual Antenna Heading) in ArduPilot 4.6 Dev

VladimirAndreev1985, 2024-06-24:
Bug report
Issue details

I am experiencing an issue with the configuration of my flight controller and RTK system. The copter only arms in Stabilize mode and fails to switch to other flight modes like AltHold with the error: "Mode change to ALT_HOLD failed: need alt estimate."

Version
ArduPilot 4.6 Dev

Platform
[ ] All
[ ] AntennaTracker
[X] Copter
[ ] Plane
[ ] Rover
[ ] Submarine

Airframe type
Quadcopter

Hardware type
Matek H743 Slim V3

RTK System
H-RTK Mosaic-H (Dual Antenna Heading) from Holybro

Issue details

Reset all settings to default values.
Applied the recommended settings from the Holybro documentation (https://docs.holybro.com/gps-and-rtk-system/h-rtk-mosaic-h/setup-and-getting-started-ardupilot).
Verified correct installation and functionality of the barometer and GPS.
Checked EKF settings:
EK3_ALT_SOURCE = 0 (Barometer)
EK3_HGT_SOURCE = 0 (Barometer)
EK3_SRC1_YAW = 2 (GPS), but also tried 1 (Compass) to avoid errors.
The problem persists even after resetting and reapplying all settings. It appears that the system is unable to obtain a reliable altitude estimate outside of Stabilize mode.

vincentpoont2, 2024-06-25:
@VladimirAndreev1985 Do you have log you can provide?

VladimirAndreev1985, 2024-06-25:
Hi @vincentpoont2 
https://filebin.net/fagcr5tzkwo9djy7

peterbarker, 2024-11-24:
@VladimirAndreev1985 unfortunately that file upload's no longer available.

Could you upload it again, please?

You didn't state in here whether the problem existed before configuring the system to use GPS-for-yaw.