ArduPilot/ardupilot #27389: Unexpected Large Target Bearing at WP1

isso11, 2024-06-25:
**Issue details**

This could be a potential issue I've encountered multiple times while using the Software In The Loop (SITL) simulation. During my tests, I observed that upon reaching the first waypoint (WP1), the system consistently outputs a "Large" target bearing value. This occurs despite the fact that the whole mission path is a straight line, and there are no significant bearing errors observed throughout the route.

**Version**
V "4.5.4" commit "e8f937aa"

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[●] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Fixed Wing

**Hardware type**
Issue found in SITL (Built-In SITL and flightaxis SITL)

**Logs**
[Log](https://drive.google.com/file/d/1QRGHz5YJZMLDWCmLjKhsr1HstNt53Rab/view?usp=sharing)

Georacer, 2024-06-26:
Indeed, in the log you provided, the target bearing "jumps" after reaching each waypoint.
However, the actual navigation bearing seems to "jump" only at the first waypoint, bringing along a jump in the roll demand.
[image]

I can't reproduce the issue on my system, though.

Can you provide the SITL command and the mission you use?

isso11, 2024-06-26:
> Indeed, in the log you provided, the target bearing "jumps" after reaching each waypoint. However, the actual navigation bearing seems to "jump" only at the first waypoint, bringing along a jump in [...]
In the Simulation tab command line: `--home=49.3137086,30.1507759,226.95,22`

Just a trivial mission, Takeoff and then Auto.

Here's the [mission](https://drive.google.com/file/d/1xkiqnbcfT-f1gZwkVNUhIU4sLW1njUud/view?usp=sharing).

Georacer, 2024-06-27:
Your mission has a "misalignment" of ~1.5deg.
And yes, with this mission it's reproducible in `master`:
[image]

I also modified your mission to "misalign" the segments by ~5 degrees and it still happens, to a reduced degree:
[image]

Here's the modified mission and the log.
[21217.zip](https://github.com/user-attachments/files/16011964/21217.zip)
[mission_modified.txt](https://github.com/user-attachments/files/16011965/mission_modified.txt)

Looks like a veritable bug to me.