ArduPilot/ardupilot #27421: Iflight Blitz H743 Wing Ralay 2 (GPIO 81) not working

qazico, 2024-06-29:
Relay 2 does not turn off 9v pin on flight controller.

Assigned CH8 to relay 2. Verified relay 2 is associated to GPIO pin 81. Tried different relay default options with no change. iFlight H743 Wing hwdef includes Blitz H743Pro hwdef. H743 Wing includes 2 camera inputs (H743 Pro only has one camera input). Suspect 2 separate GPIO required on H743 Wing (one for camera switch one for 9v vsw)

Plane 4.5.3/4.5.4
**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Plane

**Hardware type**
iFlight H743 Wing

qazico, 2024-07-02:
Only jumpers I see are for the servo voltage jumper pads. This fc is a bit different than the matek offerings.

peterbarker, 2024-08-03:
@qazico did you try setting the parameter as described in that other PR?

qazico, 2024-08-03:
Yes. I took a look at the hwdef for the H743 Wing and it includes the Blitz H743 pro hwdef. If you then take a look at the H7 pro hwdef it defines the vtx power relay. I think the problem is that the H743 wing is equipped with 2 camera inputs while the H7 pro only has one. I believe the H743 wing requires two relays, one for the camera switching and one for the vtx power. I’m guessing gpio pin 81 is assigned to the camera switch on the h743 wing.

andyp1per, 2024-08-05:
I'm checking with iFlight

jlpoltrack, 2024-09-19:
Just posting a finding in case as it seems relevant.  As there's no mention of video switching in the Docs as of 9/19, one needs to use the Vi2 pad for analog cameras.  According to INAV the GPIO to switch these things are PD10 and PD11: https://github.com/iNavFlight/inav/blob/5260d88f9c09d6d88d1e856caea62f704faed5a1/src/main/target/IFLIGHT_BLITZ_H7_PRO/target.h#L156-L160

andyp1per, 2024-09-21:
I created this - https://github.com/ArduPilot/ardupilot/pull/28185 - be great if someone could try it

jlpoltrack, 2024-09-21:
Can confirm what qazico sees on the stock firmware - that toggling Relay2 has no impact on the 9 volt output.

Andy - could you share a firmware binary for your PR?  I don't have an AP build environment set up.

andyp1per, 2024-09-21:
> Can confirm what qazico sees on the stock firmware - that toggling Relay2 has no impact on the 9 volt output.
> 
> Andy - could you share a firmware binary for your PR? I don't have an AP build environment set up.

Wing or Pro?

jlpoltrack, 2024-09-21:
> Wing or Pro?

Wing please.

andyp1per, 2024-09-23:
[arduplane.zip](https://github.com/user-attachments/files/17093652/arduplane.zip)

peterbarker, 2024-11-24:
Ping @jlpoltrack - Andy supplied a binary for you to test.

jlpoltrack, 2024-11-24:
@andyp1per @peterbarker 

Few notes:

- Andy's PR does work - GPIO 82 switches the camera input, tested this morning.
- I don't see hardware wise how it is possible disable the 9V output, there's no GPIO routed to the PDB to enable / disable the regulator.  See photo:
[image]
- The CAN_L / CAN_H pads are swapped.

closed 2025-01-25 by peterbarker