ArduPilot/ardupilot #27474: MAVLink signing active also for USB, in contrast to docs/desired behavior?

olliw42, 2024-07-06:
## Bug report

I may missunderstand the docs (https://ardupilot.org/copter/docs/common-MAVLink2-signing.html) and how it's all supposed to work, but it seems to me that MAVLink signing is active also when connected via the USB port.

**Issue details**

I do have a MatekH743 flight controller connected via USB to the PC, as well as via a USB-TTL adapter on SERIAL2 which is enumerated as COM23. I can connect to the FC via MissionPlanner both via the USB or SERIAL2.
When connected via the USB, following the docs, I can successfully enable MAVLink signing, and in the Stats pane I do see the Signing checkbox marked (for when connected via USB). Repowering and reconnecting via USB works fine, and the signing checkbox is again seen. Now connecting via SERIAL2l on COM23 also works fine, and also the signing checkbox is marked in the MissionPlanner Stats pane. I can do that as often as I want. So, to me, it looks like I have set up things correctly.

I now delete the authkeys.xml, to simulate the case of a GCS which doesn't know the signing key. From there on I cannot connect to the FC anymore, whatever I do. Also not via USB.

The only way to recover I figured out yet is to flash the firmware for a different vehicle, to reset the params.

**Version**
ArduPilot 4.6, Copter or Plane vehicles
MissionPlanner 1.3.81

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
-

**Hardware type**
MatekH743 slim

**Logs**
-

peterbarker, 2024-07-10:
I'm unable to replicate this on SITL.

Started SITL like this:
```
./Tools/autotest/sim_vehicle.py -v ArduCopter --gdb --debug --no-mavproxy
```

Connected MAVProxy to 5760 like this:
```
mavproxy.py --master tcp:localhost:5760
```
set up signing like this:
```
STABILIZE> signing key bobbobbob
STABILIZE> Setup signing key
```

Disconnected and reconnected on that port, can communicate with autopilot without setting up the signing key.

Then connected to port 5762 like this:
```
pbarker@fx:~$ mavproxy.py --master tcp:localhost:5762
```
... and it can receive traffic but not have traffic to the autopilot be accepted.  I need to 
```
STABILIZE> signing key bobbobbob
STABILIZE> Setup signing key

STABILIZE> AP: ArduCopter V4.6.0-dev (20257125)
AP: 40338c121ecd45b0bb475d0700199847
AP: Frame: QUAD/PLUS
Received 1386 parameters
```
to get it to take effect

Will try on real hardware.

peterbarker, 2024-07-10:
Crap, did I just leak my super-secret key?

peterbarker, 2024-07-10:
.... I've just tested on a Pixhawk6X here and can't replicate what you're seeing.

USB always works regardless of signing setup.

SERIAL1 doesn't permit traffic in when signing is active unless you set the signing key in the MAVProxy connected to that port.

I wonder if this is a MissionPlanner issue rather than an ArduPilot issue somehow?

Could you do some testing with MAVProxy, please?

olliw42, 2024-07-10:
many thanks for looking at it
I can't do testing with mavproxy, that's beyond my paygrade.

I was concluding that it should be an ArduPilot issue since I would think accepting the message on USB happens on ArduPilot side. But what do I know.

I will try to make a video maybe, or offer more. May take few days.

peterbarker, 2024-08-13:
@olliw42 are you still looking at this one?

Just re-read your problem description.

> I now delete the authkeys.xml, to simulate the case of a GCS which doesn't know the signing key. From there on I cannot connect to the FC anymore, whatever I do. Also not via USB.

`authkeys.xml` is a MissionPlanner thing, I assume?

I can't see how removing that would stop you being able to connect to the autopilot - that is very strange indeed!  You won't be able to fetch parameters, but definitely should still get a mavlink stream if there's one there to be had.  Remember that if you don't have keys you can't send a packet to the autopilot which will start it streaming in any way, so you do need to have the autopilot already streaming somehow.

peterbarker, 2025-05-11:
Ping @olliw42 on this one

olliw42, 2025-05-11:
thx for asking
I unfortunately do not have the time to check this again.
So, I'm closing,

closed 2025-05-11 by olliw42