ArduPilot/ardupilot #27528: Custom Build failed

twistedwings, 2024-07-11:
## Bug report

Custom Build on Custom Build Server failed.

**Issue details**
see log file

**Platform**
[  ] All
[  ] AntennaTracker
[X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Matek F405-CTR

**Logs**

https://custom.ardupilot.org/builds/copter:MatekF405:d9518fff3c8588d1d3264cb60818d88f51cb84b0:843fdf16f9caf3173569d8310748444d/build.log

IamPete1, 2024-07-12:
Sorry, that build has fallen off the end of the cache. Can you reproduce and upload the log here please.

twistedwings, 2024-07-12:
I am not sure if it's exactly the same config.

[Custom Build.txt](https://github.com/user-attachments/files/16193902/Custom.Build.txt)

IamPete1, 2024-07-12:
The error is because you have enabled DroneCAN but the MatekF405 does have CAN hardware.

IamPete1, 2024-07-12:
I have opened a issue on for the custom firmware server, it should not allow things to be selected for which the board does not have hardware support. 

https://github.com/ArduPilot/CustomBuild/issues

closed 2024-07-12 by IamPete1