ArduPilot/ardupilot #27549: Pitch limit for landing

Mahdikh7, 2024-07-15:
hi
I want the pitch to be higher than -20 when the drone lands, for example up to -30 because the landing site is in a pit.
Thank you for telling me the location of the code that is stored in the source -20
I even set the minimum PITCH limit parameter to -30, but it didn’t go further than -20
If you tell me exactly in which file of the source code this PITCH limit is placed, I can re-create the firmware and install it on Pixhawk.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[*] Plane
[  ] Rover
[  ] Submarine

magicrub, 2024-07-15:
It's more than just about pitch. The problem is you're overspeeding. Raise your airspeed max

Mahdikh7, 2024-07-19:
> It's more than just about pitch. The problem is you're overspeeding. Raise your airspeed max

Well, I am doing this test in the simulator, and the speed does not exceed a certain limit
That's why I need to find the location of the codes that write this restriction

mduclehcm, 2024-07-22:
@Mahdikh7 
[image]
The current maximum range for landing pitch is from -5 to 40 degrees. However, TECS only utilizes the landing pitch maximum value after flare. Therefore, ensure that the landing flare parameter is set correctly.