ArduPilot/ardupilot #27591: COPTER: Could not compile AP v 4.6-dev for SpeedyBee F4 v3

knoppixmeister, 2024-07-18:
Could not compile AP copter and plane version of 4.6-dev branch for SpeedyBee F4 v3

```
No target_list file found, creating
Generating compile_commands.json
Build commands will be stored in build/speedybeef4v3/compile_commands.json
Generating compile_commands.json
Build commands will be stored in build/speedybeef4v3/compile_commands.json
[ 476/1002] Compiling libraries/AP_Notify/GPIO_LED_1.cpp
[ 498/1002] Compiling libraries/AP_OpenDroneID/AP_OpenDroneID_DroneCAN.cpp
[ 499/1002] Compiling libraries/AP_OpenDroneID/AP_OpenDroneID.cpp
[ 500/1002] Compiling libraries/AP_OpticalFlow/AP_OpticalFlow_UPFLOW.cpp
[ 501/1002] Compiling libraries/AP_OpticalFlow/AP_OpticalFlow_SITL.cpp
[ 502/1002] Compiling libraries/AP_OpticalFlow/AP_OpticalFlow_Pixart.cpp
[ 503/1002] Compiling libraries/AP_OpticalFlow/AP_OpticalFlow_PX4Flow.cpp
[ 504/1002] Compiling libraries/AP_OpticalFlow/AP_OpticalFlow_Onboard.cpp
[ 505/1002] Compiling libraries/AP_OpticalFlow/AP_OpticalFlow_MSP.cpp
[ 506/1002] Compiling libraries/AP_OpticalFlow/AP_OpticalFlow_HereFlow.cpp
[ 507/1002] Compiling libraries/AP_OpticalFlow/AP_OpticalFlow_Calibrator.cpp
[ 508/1002] Compiling libraries/AP_OpticalFlow/AP_OpticalFlow.cpp
[ 509/1002] Compiling libraries/AP_OpticalFlow/AP_OpticalFlow_MAV.cpp
[ 510/1002] Compiling libraries/AP_OpticalFlow/AP_OpticalFlow_Backend.cpp
[ 511/1002] Compiling libraries/AP_OpticalFlow/AP_OpticalFlow_CXOF.cpp
../../libraries/AP_Notify/GPIO_LED_1.cpp:26:2: error: #error "define AP_NOTIFY_GPIO_LED_A_PIN"
   26 | #error "define AP_NOTIFY_GPIO_LED_A_PIN"
      |  ^~~~~
compilation terminated due to -Wfatal-errors.

Waf: Leaving directory `/Users/knoppix/Desktop/ardupilot/build/speedybeef4v3'
Build failed
 -> task in 'objs/AP_Notify' failed (exit status 1): 
        {task 4369509152: cxx GPIO_LED_1.cpp -> GPIO_LED_1.cpp.0.o}
 (run with -v to display more information)

```

**Version**
4.6-dev

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Copter & Plane

**Hardware type**
SpeedyBee F4 v3

**Logs**

andyp1per, 2024-07-18:
Looks like your change @peterbarker

peterbarker, 2024-07-31:
This has been fixed in master a few days ago.  Apologies for the breakage.

closed 2024-07-31 by peterbarker