ArduPilot/ardupilot #27603: Add support for Spektrum Smart ESC (and battery)

iksaif, 2024-07-21:
## Feature request

**Is your feature request related to a problem? Please describe.**
Spektrum Smart ESC and batteries export detailed telemetry (https://www.horizonhobby.com/on/demandware.static/Sites-horizon-us-Site/Sites-horizon-master/default/Manuals/Avian-Manual-EN.pdf)

```
SMART Throttle*:

Spektrum™ Avian™ ESCs are compatible with SMART Throttle.
SMART Throttle combines throttle signals with telemetry data from
the ESC on one normal three-wire servo connector. SMART Throttle
compatible receivers will automatically detect a SMART Throttle
ESC and begin to send telemetry information to your transmitter.
Using the SMART Throttle connection, this ESC can send
voltage current and other telemetry data. It can also pass along
battery data from compatible Spektrum SMART batteries. A
Spektrum SMART battery with an IC™ connector is required for
battery data. EC™ connectors are compatible for basic operation, but
will not provide SMART battery data.

Only certain Spektrum telemetry receivers include SMART
Throttle; check your receiver manual for more information. If the
ESC is not connected to a SMART Throttle compatible receiver,
no telemetry data from the ESC will be available, but the ESC
will operate normally with a common servo signal (PWM).
*A Spektrum DSMX transmitter with telemetry is required for
SMART Throttle. 
```

https://www.spektrumrc.com/on/demandware.static/-/Sites-horizon-master/default/dwbbc4f00a/Manuals/SPM-Firma-Avian-ESC-Chart-Update.pdf implies that this is done through SRXL2. This also seems to mean that if we read the telemetry we need to use SRXL2 to control the ESC ().

**Describe the solution you'd like**

This to be supported to be able to read telemetry from a Smart ESC and Battery (like https://ardupilot.org/copter/docs/common-fettec-onewire.html#common-fettec-onewire)

**Describe alternatives you've considered**

N/A

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**

Related issue https://github.com/SpektrumRC/SRXL2/issues/3

andyp1per, 2024-07-21:
I agree it would be nice. Really need HH to show some ArduPilot love to make this happen as I don't have hardware or time.

iksaif, 2024-07-22:
I have the hardware but not a lot of time these day. I might do some experiments if I get some answers in https://github.com/SpektrumRC/SRXL2/issues/3 (in particular for the protocol and wiring ..)

peterbarker, 2024-08-13:
More information here: https://discuss.ardupilot.org/t/spektrum-smart-escs-and-srxl2/95104/5

iksaif, 2024-08-16:
This is useful, we'd need a bit more doc from Spektrum  to implement all of that.