ArduPilot/ardupilot #27604: PLANE: During a Loiter mission, the throttle decreased to 0 for no apparent reason

bbc25185, 2024-07-21:
## Bug report

**Issue details**

PLANE:
Twice during the same flight, while in loiter flight mode , the throttle autonomously decided to drop almost to 0.
In the first case he was taken back to FBWA, brought back to altitude and put back in Loiter.
In the second case he even landed with the throttle at 0, while he was in Loiter.
I tried looking to see if there was some setting or condition that caused it, but I couldn't find anything.
Thanks in advance for any response.

**Version**
Arduplane 4.5.3

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
flying wing

**Hardware type**
Matek H743 Wlite with BDshot firmware

**Logs**
[2024-06-18 06-37-15.anon.zip](https://github.com/user-attachments/files/16324791/2024-06-18.06-37-15.anon.zip)

IamPete1, 2024-07-21:
This is the elevator output:

[image]

It ran out of up elevator, the throttle cut was in response to gaining airspeed because of the drop in pitch. 

Not sure on the cause, you can see its using increasingly more up elevator through out the flight. 

[image]

What is a little unexpected is that once it did recover the elevator seemed to reset and then start increasing again. 

We need some more information about the vehicle to get much further I think. It would also be a good idea to give it a thorough check over for hardware problems.

bbc25185, 2024-07-21:
The vehicle has nothing obvious that isn't working (also because, after taking it back in FBWA the first time, we flew for almost 30 minutes without problems). As mentioned, the plane landed smoothly and further flights were made (although not as long) and there were no other problems!
The vehicle is a simple flying wing with 3 simple channels (throttle, right elevon, left elevon).
If you make the comparison, you notice that the GPS speed of the vehicle was not high... on the contrary... it was quite low when it started to reduce the throttle... and when it was picked up for the first time, it was at the speed of 11m/s (always relative to the ground).
Furthermore, if you look at the GPS speed, you notice a slight slowdown in the average speed which could be a consequence of that elevator command!
[image]

peterbarker, 2024-08-03:
> EDIT: can be AUTOTRIM options enabled the cause of this???

If there's a bug, yes.

It was not extremely active in this flight, it was not the cause.

peterbarker, 2024-08-03:
> What is a little unexpected is that once it did recover the elevator seemed to reset and then start increasing again.

The PID was probably reset as it went through FBWA mode.