ArduPilot/ardupilot #27608: Copter: reboot with MP will load for a long time when set CAN_P1_DRIVER to 1

cuav-chen2, 2024-07-22:
## Bug report

**Issue details**
In current master branch if compile with `copter`, when set `CAN_P1_DRIVER ` or `CAN_P2_DRIVER ` to `1` and send a reboot command to autopilot, Mission Planner will reconnect automaticlly and enter a infinite loading state just like that:
[image]

I' ve tested in CUAV-X7 and CUAVv5Nano autopilot and both had this issue. To solve this loading problem, you must disconnect and reconnect manually in MP, or click the `Retry Now` button in `CONFIG` tabs. When MP staying in loading state, the console window shows that it is reading a 'None' param. I guess it has something to do with the `@PARAM/param.pack` file downloaded by mavlink on connect.
[image]

I'm not sure if this is a MP or Ardupilot issue. It seems happened after this PR https://github.com/ArduPilot/ardupilot/pull/26497 .

**Version**
ArduCopter V4.6.0-dev
Mission Planner 1.3.81 build 1.3.8741.25556

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Logs**
no logs.

rmackay9, 2024-07-23:
Hi @cuav-chen2,

Thanks very much for the report.  I suspect (but don't know for sure) that this is a [Mission Planner issue](https://github.com/ArduPilot/MissionPlanner/issues)

rmackay9, 2024-07-23:
Hi @cuav-chen2,

Thanks very much for doing that additional testing.  We probably need to re-raise this issue over on the MP issues list then

cuav-chen2, 2024-07-23:
Sure, I'll do that later.