ArduPilot/ardupilot #27632: ublox GPS reports fix type of 4

Baudouin0, 2024-07-24:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)

ublox GPS reports fix type of 4 - this is GNSS+dead reckoning as the puck contains acceleromters.
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

Debug terminal reports 'Unexpected state 33 or 35'

**Version**
4.5.4

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
F450

**Hardware type**
PixHawk 2.4.8

**Logs**
Unexpected state 33.

Baudouin0, 2024-07-24:
AP_GPS_UBLOX.cpp:
Make case 3 and 4 the same.

        switch (_buffer.pvt.fix_type)
        {
            case 0:
                state.status = AP_GPS::NO_FIX;
                break;
            case 1:
                state.status = AP_GPS::NO_FIX;
                break;
            case 2:
                state.status = AP_GPS::GPS_OK_FIX_2D;
                break;
            case 3:
            case 4:
                state.status = AP_GPS::GPS_OK_FIX_3D;
                if (_buffer.pvt.flags & 0b00000010)  // diffsoln
                    state.status = AP_GPS::GPS_OK_FIX_3D_DGPS;
                if (_buffer.pvt.flags & 0b01000000)  // carrsoln - float
                    state.status = AP_GPS::GPS_OK_FIX_3D_RTK_FLOAT;
                if (_buffer.pvt.flags & 0b10000000)  // carrsoln - fixed
                    state.status = AP_GPS::GPS_OK_FIX_3D_RTK_FIXED;
                break;
            case 5:
                state.status = AP_GPS::NO_FIX;
                break;
            default:
                state.status = AP_GPS::NO_FIX;
                break;
        }

peterbarker, 2024-07-25:
Which physical device is this?

The patch you propose here looks like it's probably correct.  Did you want to create a PR for it?

Baudouin0, 2024-07-26:
> CAUTION: This email originated from outside of the organisation. Do not reply, click links or open attachments unless you recognise the sender's email address and know the content is safe. If you have any doubts contact the IT ServiceDesk.
>
> [EXTERNAL] This email was sent from outside of the Codan Group. Use caution when opening attachments or links from unknown senders.

It’s a Ublox M8N with built in compass.
NEO-M8 series | u-blox<https://www.u-blox.com/en/product/neo-m8-series>

Should be able to do pull request for you. I have built the code and retested for Pixhawk 2.4.8.

Charles Baudouin
Chief Technical Architect

Email: ***@***.******@***.***>
Mobile / WhatsApp: +44 7973 146 671
Telephone: +44 1489 884 430
Skype / Teams: ***@***.***

DOMO TACTICAL COMMUNICATIONS (DTC) LTD TRADING AS DTC
Registered office Fusion 2, 1100 Parkway, Whiteley, Hampshire, PO15 7AB, United Kingdom
Registered in England and Wales, company number 1456922
[cid:dtclogo203x88_8fe7e5cd-210b-49fc-b334-f34189be7d63.png]
[cid:dtcbanner310x28_238a1048-71d6-4715-a939-5711d6112998.png]<https://www.domotactical.com> [cid:dtclinkedin_7ede4504-e8a4-4622-8c8d-cd35b8a00736.png] <https://www.linkedin.com/company/domo-tactical-communications-dtc-ltd/>   [cid:dtctwitter_4b884fb4-cf7b-4376-8ca7-c6d300a80c2f.png] <https://twitter.com/domobroadcast>   [cid:dtcyoutube_8642fa77-2f6f-494f-bb70-281557d074b8.png] <https://www.youtube.com/channel/UC0nBUhsZ9iakjODp9nYRDSw>

[cid:dtcemailsignature-01_debbf482-cca7-4406-98df-a0314484b498.png]

PLEASE CONSIDER THE ENVIRONMENT BEFORE PRINTING THIS EMAIL
This e-mail and any files transmitted with it is intended solely for the addressee(s) and may contain information that is proprietary, privileged, company confidential and/or exempt from disclosure under applicable law. If you are not the intended addressee(s), any disclosure, reproduction, copying, distribution or other use of the E-mail is strictly prohibited. In the event this document(s) contains technical data within the definition of the International Traffic in Arms Regulations, it is subject to the export control laws of the U.S. Government. Transfer of this data by any means to a foreign person, whether in the United States or abroad, without an export license or other approval from the U.S. Department of State, is prohibited. Neither the company nor any subsidiary, affiliate or associated company nor any individual sending this E-mail accepts any liability in respect of the content (including errors and omissions).

peterbarker, 2024-07-26:
I don't think uBlox ship M8 devices with compasses. They're often integrated by other parties.

Please link to the specific device which is sending these messages.

Baudouin0, 2024-07-26:
As you said assembled by a third party and then branded. The main thing is the M8N contains some dead reckoning software and sensors.

Ublox NEO-M8N GPS With Compass - Unmanned Tech Shop<https://www.unmannedtechshop.co.uk/product/ublox-neo-m8n-gps-with-compass/>

Charles Baudouin
Chief Technical Architect

Email: ***@***.******@***.***>
Mobile / WhatsApp: +44 7973 146 671
Telephone: +44 1489 884 430
Skype / Teams: ***@***.***

DOMO TACTICAL COMMUNICATIONS (DTC) LTD TRADING AS DTC
Registered office Fusion 2, 1100 Parkway, Whiteley, Hampshire, PO15 7AB, United Kingdom
Registered in England and Wales, company number 1456922
[cid:dtclogo203x88_8fe7e5cd-210b-49fc-b334-f34189be7d63.png]
[cid:dtcbanner310x28_238a1048-71d6-4715-a939-5711d6112998.png]<https://www.domotactical.com> [cid:dtclinkedin_7ede4504-e8a4-4622-8c8d-cd35b8a00736.png] <https://www.linkedin.com/company/domo-tactical-communications-dtc-ltd/>   [cid:dtctwitter_4b884fb4-cf7b-4376-8ca7-c6d300a80c2f.png] <https://twitter.com/domobroadcast>   [cid:dtcyoutube_8642fa77-2f6f-494f-bb70-281557d074b8.png] <https://www.youtube.com/channel/UC0nBUhsZ9iakjODp9nYRDSw>

[cid:dtcemailsignature-01_debbf482-cca7-4406-98df-a0314484b498.png]

PLEASE CONSIDER THE ENVIRONMENT BEFORE PRINTING THIS EMAIL
This e-mail and any files transmitted with it is intended solely for the addressee(s) and may contain information that is proprietary, privileged, company confidential and/or exempt from disclosure under applicable law. If you are not the intended addressee(s), any disclosure, reproduction, copying, distribution or other use of the E-mail is strictly prohibited. In the event this document(s) contains technical data within the definition of the International Traffic in Arms Regulations, it is subject to the export control laws of the U.S. Government. Transfer of this data by any means to a foreign person, whether in the United States or abroad, without an export license or other approval from the U.S. Department of State, is prohibited. Neither the company nor any subsidiary, affiliate or associated company nor any individual sending this E-mail accepts any liability in respect of the content (including errors and omissions).

peterbarker, 2024-07-26:
> As you said assembled by a third party and then branded. The main thing is the M8N contains some dead reckoning software and sensors. Ublox NEO-M8N GPS With Compass - Unmanned Tech Shop<https://www.unmannedtechshop.co.uk/product/ublox-

I'm pretty sure that device simply has a compass available over that 6-pin connector - i.e. not integrated with the M8.  It's not an uncommon setup.

You also mentioned accels and gyros - and they're definitely not present.

Are you saying that *specific* product causes the error you are seeing?  I'll need to get one to verify.

Baudouin0, 2024-07-26:
Sorry our company prevents pushing back onto git.
Here’s the patch:

diff --git a/libraries/AP_GPS/AP_GPS_UBLOX.cpp b/libraries/AP_GPS/AP_GPS_UBLOX.cpp
index ac947b7a37..198b673c3b 100644
--- a/libraries/AP_GPS/AP_GPS_UBLOX.cpp
+++ b/libraries/AP_GPS/AP_GPS_UBLOX.cpp
@@ -1675,6 +1675,7 @@ AP_GPS_UBLOX::_parse_gps(void)
                 state.status = AP_GPS::GPS_OK_FIX_2D;
                 break;
             case 3:
+            case 4:
                 state.status = AP_GPS::GPS_OK_FIX_3D;
                 if (_buffer.pvt.flags & 0b00000010)  // diffsoln
                     state.status = AP_GPS::GPS_OK_FIX_3D_DGPS;
@@ -1683,11 +1684,6 @@ AP_GPS_UBLOX::_parse_gps(void)
                 if (_buffer.pvt.flags & 0b10000000)  // carrsoln - fixed
                     state.status = AP_GPS::GPS_OK_FIX_3D_RTK_FIXED;
                 break;
-            case 4:
-                GCS_SEND_TEXT(MAV_SEVERITY_INFO,
-                                "Unexpected state %d", _buffer.pvt.flags);
-                state.status = AP_GPS::GPS_OK_FIX_3D;
-                break;
             case 5:
                 state.status = AP_GPS::NO_FIX;
                 break;

Charles Baudouin
Chief Technical Architect

Email: ***@***.******@***.***>
Mobile / WhatsApp: +44 7973 146 671
Telephone: +44 1489 884 430
Skype / Teams: ***@***.***

DOMO TACTICAL COMMUNICATIONS (DTC) LTD TRADING AS DTC
Registered office Fusion 2, 1100 Parkway, Whiteley, Hampshire, PO15 7AB, United Kingdom
Registered in England and Wales, company number 1456922
[cid:dtclogo203x88_8fe7e5cd-210b-49fc-b334-f34189be7d63.png]
[cid:dtcbanner310x28_238a1048-71d6-4715-a939-5711d6112998.png]<https://www.domotactical.com> [cid:dtclinkedin_7ede4504-e8a4-4622-8c8d-cd35b8a00736.png] <https://www.linkedin.com/company/domo-tactical-communications-dtc-ltd/>   [cid:dtctwitter_4b884fb4-cf7b-4376-8ca7-c6d300a80c2f.png] <https://twitter.com/domobroadcast>   [cid:dtcyoutube_8642fa77-2f6f-494f-bb70-281557d074b8.png] <https://www.youtube.com/channel/UC0nBUhsZ9iakjODp9nYRDSw>

[cid:dtcemailsignature-01_debbf482-cca7-4406-98df-a0314484b498.png]

PLEASE CONSIDER THE ENVIRONMENT BEFORE PRINTING THIS EMAIL
This e-mail and any files transmitted with it is intended solely for the addressee(s) and may contain information that is proprietary, privileged, company confidential and/or exempt from disclosure under applicable law. If you are not the intended addressee(s), any disclosure, reproduction, copying, distribution or other use of the E-mail is strictly prohibited. In the event this document(s) contains technical data within the definition of the International Traffic in Arms Regulations, it is subject to the export control laws of the U.S. Government. Transfer of this data by any means to a foreign person, whether in the United States or abroad, without an export license or other approval from the U.S. Department of State, is prohibited. Neither the company nor any subsidiary, affiliate or associated company nor any individual sending this E-mail accepts any liability in respect of the content (including errors and omissions).

peterbarker, 2024-07-30:
From the f9p datasheet (https://cdn.sparkfun.com/assets/learn_tutorials/8/5/6/ZED-F9P_UBX_NMEA_and_RTCM_protocols.pdf):
[image]

I still want to know which device emits this value in this field.  I've asked for confirmation that the above device does.

Baudouin0, 2024-07-30:
When I've connected it back up I will see what the MSG_MON_VER is. I cannot see into the unit as the puck is opaque.

GCS_SEND_TEXT(MAV_SEVERITY_INFO,
                                             "u-blox %s%s%d HW: %s SW: %s",
                                             _module, mod != nullptr ? " " : "",
                                             state.instance + 1,
                                             _version.hwVersion,
                                             _version.swVersion);

Charles Baudouin
Chief Technical Architect

Email: ***@***.******@***.***>
Mobile / WhatsApp: +44 7973 146 671
Telephone: +44 1489 884 430
Skype / Teams: ***@***.***

DOMO TACTICAL COMMUNICATIONS (DTC) LTD TRADING AS DTC
Registered office Fusion 2, 1100 Parkway, Whiteley, Hampshire, PO15 7AB, United Kingdom
Registered in England and Wales, company number 1456922
[cid:dtclogo203x88_8fe7e5cd-210b-49fc-b334-f34189be7d63.png]
[cid:dtcbanner310x28_238a1048-71d6-4715-a939-5711d6112998.png]<https://www.domotactical.com> [cid:dtclinkedin_7ede4504-e8a4-4622-8c8d-cd35b8a00736.png] <https://www.linkedin.com/company/domo-tactical-communications-dtc-ltd/>   [cid:dtctwitter_4b884fb4-cf7b-4376-8ca7-c6d300a80c2f.png] <https://twitter.com/domobroadcast>   [cid:dtcyoutube_8642fa77-2f6f-494f-bb70-281557d074b8.png] <https://www.youtube.com/channel/UC0nBUhsZ9iakjODp9nYRDSw>

[cid:dtcemailsignature-01_debbf482-cca7-4406-98df-a0314484b498.png]

PLEASE CONSIDER THE ENVIRONMENT BEFORE PRINTING THIS EMAIL
This e-mail and any files transmitted with it is intended solely for the addressee(s) and may contain information that is proprietary, privileged, company confidential and/or exempt from disclosure under applicable law. If you are not the intended addressee(s), any disclosure, reproduction, copying, distribution or other use of the E-mail is strictly prohibited. In the event this document(s) contains technical data within the definition of the International Traffic in Arms Regulations, it is subject to the export control laws of the U.S. Government. Transfer of this data by any means to a foreign person, whether in the United States or abroad, without an export license or other approval from the U.S. Department of State, is prohibited. Neither the company nor any subsidiary, affiliate or associated company nor any individual sending this E-mail accepts any liability in respect of the content (including errors and omissions).

From the f9p datasheet (https://cdn.sparkfun.com/assets/learn_tutorials/8/5/6/ZED-F9P_UBX_NMEA_and_RTCM_protocols.pdf<https://cdn.sparkfun.com/assets/learn_tutorials/8/5/6/ZED-F9P_UBX_NMEA_and_RTCM_protocols.pdf>):
image.png (view on web)<https://github.com/user-attachments/assets/7500c4e8-82b0-44e1-a8eb-b9134832017c>

Baudouin0, 2024-08-02:
Sorry for the delay.
The GPS device is

ublox NEO-M8L- 1 HW: 00080000 SW: EXT CORE 3.01 (1ec93f)

NEO-M8L series | u-blox<https://www.u-blox.com/en/product/neo-m8l-series>

Charles.

Charles Baudouin
Chief Technical Architect

Email: ***@***.******@***.***>
Mobile / WhatsApp: +44 7973 146 671
Telephone: +44 1489 884 430
Skype / Teams: ***@***.***

DOMO TACTICAL COMMUNICATIONS (DTC) LTD TRADING AS DTC
Registered office Fusion 2, 1100 Parkway, Whiteley, Hampshire, PO15 7AB, United Kingdom
Registered in England and Wales, company number 1456922
[cid:dtclogo203x88_8fe7e5cd-210b-49fc-b334-f34189be7d63.png]
[cid:dtcbanner310x28_238a1048-71d6-4715-a939-5711d6112998.png]<https://www.domotactical.com> [cid:dtclinkedin_7ede4504-e8a4-4622-8c8d-cd35b8a00736.png] <https://www.linkedin.com/company/domo-tactical-communications-dtc-ltd/>   [cid:dtctwitter_4b884fb4-cf7b-4376-8ca7-c6d300a80c2f.png] <https://twitter.com/domobroadcast>   [cid:dtcyoutube_8642fa77-2f6f-494f-bb70-281557d074b8.png] <https://www.youtube.com/channel/UC0nBUhsZ9iakjODp9nYRDSw>

[cid:dtcemailsignature-01_debbf482-cca7-4406-98df-a0314484b498.png]

PLEASE CONSIDER THE ENVIRONMENT BEFORE PRINTING THIS EMAIL
This e-mail and any files transmitted with it is intended solely for the addressee(s) and may contain information that is proprietary, privileged, company confidential and/or exempt from disclosure under applicable law. If you are not the intended addressee(s), any disclosure, reproduction, copying, distribution or other use of the E-mail is strictly prohibited. In the event this document(s) contains technical data within the definition of the International Traffic in Arms Regulations, it is subject to the export control laws of the U.S. Government. Transfer of this data by any means to a foreign person, whether in the United States or abroad, without an export license or other approval from the U.S. Department of State, is prohibited. Neither the company nor any subsidiary, affiliate or associated company nor any individual sending this E-mail accepts any liability in respect of the content (including errors and omissions).

peterbarker, 2024-08-13:
> Sorry for the delay. The GPS device is ublox NEO-M8L- 1 HW: 00080000 SW: EXT CORE 3.01 (1ec93f) NEO-M8L series | u-blox<https://www.u-blox.com/en/product/neo-m8l-series> Charles. Charles Baudouin Chief Technical Architect Email: 

Ah, right.  *That* makes sense now.

I'm guessing this GPS you're using might be connected to something else which is providing additional information to the unit sufficient for it to provide an inertial solution, thus the type-4 from it.

I've created https://github.com/ArduPilot/ardupilot/pull/27825 to move this along - we'll discuss at DevCallEU tomorow.  Note my concern on that PR - we may end up having to reject things with that fix-type rather than treat it as fix-type 3.  Anything providing a gnss + dead-reckoning solution should *probably* be coming in via AP_ExternalAHRS, rather than coming in as a GPS.