ArduPilot/ardupilot #27648: RELAY set to MAIN (IOMCU) pin doesn't turn off on versions 4.5.0 and later

LEGIONBOSS, 2024-07-25:
## Bug report

**Issue details**
When I set up a MAIN pin as a RELAY on an Orange Cube running ArduCopter 4.5.4, it turns ON when instructed, then gets stuck ON when I try to turn it OFF. This issue is not present on AUX (FMU) pins.

Steps to re-create the issue

On Copter 4.4.4 and older:
1. Set SERVO1_FUNCTION to -1 (GPIO)
2. Set RELAY1_PIN to 101 (MainOut1)
3. Reboot the board

On Copter 4.5.0 and later:
1. Set SERVO1_FUNCTION to -1 (GPIO)
2. Set RELAY1_FUNCTION to 1 (Relay)
4. Refresh the parameter list to reveal the one below
5. Set RELAY1_PIN to 101 (MainOut1)
6. Reboot the board

On both, after setting the parameters:
1. Monitor the DC voltage between MAIN OUT 1 Signal (S) and GND (-) lines
2. Toggle the relay (Relay0 in Mission Planner, if RELAY1 parameters were set) using the High and Low buttons on the Servo/Relay tab on the DATA screen in Mission Planner (the Toggle button is not a toggle really, but a Pulse)

- On Copter 4.4.4 and older, the Relay works as intended, toggles between ~0 and ~3.3 volts
- On Copter 4.5.0 and later, the Relay only turns on, but stays on when instructed to turn off

**Version**
4.5.4

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Nothing, just 

**Hardware type**
Cube Orange

**Logs**

andyp1per, 2024-07-25:
I tried this on a cube orange plus, does not work for me on either 4.4 or 4.5 @IamPete1

IamPete1, 2024-07-28:
> I tried this on a cube orange plus, does not work for me on either 4.4 or 4.5 @IamPete1

I have tested here and it works on 4.4.4 it works on CubeOrange but not a CubeOrangePlus or Pixhawk1. I'm not sure why that would be, as far as I know the IOMCU config should be the same. In anycase, it seems a little pointless to get into that too much on 4.4 since 4.5 will be different. Do you have a CubeOrange (not plus to test?)

IamPete1, 2024-07-28:
OK, I have it working again on CubeOrange, its actually not directly to do with the IOMCU. Relay started checking the pin before setting and only setting if there is a change. I never implemented get for the IOMCU, so it always returns false. Relay is trying to turn off so it thinks it can skip the set because its false already. 

However, that does not explain what is going on with the other boards.

LEGIONBOSS, 2024-07-29:
> I have tested here and it works on 4.4.4 it works on CubeOrange but not a CubeOrangePlus or Pixhawk1. I'm not sure why that would be, as far as I know the IOMCU config should be the same. In [...]

I did the mentioned tests on a CubeOrange. I only tested with Copter 4.5.4 on a CubeOrangePlus initially to check that it wasn't a hardware error.

closed 2024-07-31 by tridge