ArduPilot/ardupilot #27746: SOME ESC TELEMETRY DATA IS RECEIVED BUT NOT DISPLAYED IN OSD

LanoAstur, 2024-08-03:
## Bug report

**Issue details**

I upgraded from v4.4.0 to v4.5.5 and I find the following situation. With v4.4.0 I was receiving **in OSD** telemetry from the ESC (rpm, temperature) but with v.4.5.5 the **ESC temperature telemetry was lost** (rpm still works).

I note that the issue is in OSD because from the **STATUS tab of Mission Planner** the **ESC1_TEMP values are read correctly.**

In my case I use BLHeli32 telemetry and I don’t use BDShot.

Thanks in advance

**Version**
v4.5.5

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Glider

**Hardware type**
Mateksys F765-WING

_

andyp1per, 2024-08-05:
The temperature and RPM are now picked from the most active ESC. If you absolutely want to use ESC 1 then please set OSDx_ESC_IDX to 1

LanoAstur, 2024-08-05:
The point is that **I only have one ESC**. The ESCRPM values are displayed correctly on OSD but not the ESCTEMP values.

In my case it is not a question of which ESC I want to display information about, but **to display information about the only ESC I have**.

k-cyril, 2024-08-06:
Can confirm. The ESC temps are not displayed on the OSD at all. No matter whether ESC_IDX is set or not. Neither in 4.5 nor in 4.6 (latest). Works just fine in 4.4, and it's annoying to downgrade your FC just to debug the ESC overheating issues.

closed 2024-08-07 by tridge