ArduPilot/ardupilot #27787: AP_Baro: Support MS5837-02BA

ES-Alexander, 2024-08-07:
## Feature request

**Is your feature request related to a problem? Please describe.**
There are multiple MS5837 sensor types available, but currently only the [`-30BA`](https://www.te.com/commerce/DocumentDelivery/DDEController?Action=showdoc&DocId=Data+Sheet%7FMS5837-30BA%7FB1%7Fpdf%7FEnglish%7FENG_DS_MS5837-30BA_B1.pdf%7FCAT-BLPS0017) variant is supported, which limits which products can be used (especially since there are commercial products available for the [`-02BA`](https://www.te.com/commerce/DocumentDelivery/DDEController?Action=srchrtrv&DocNm=MS5837-02BA01&DocType=DS&DocLang=English) variant, and it's more useful for shallow applications than its counterpart).

**Describe the solution you'd like**
From the datasheets, the MS5837 sensors (and possibly multiple other MS5xxx sensor types?) seem to specify a product ID at the start of the PROM, in bits 11-5 of Word 0. Using that to differentiate between the devices should then allow a straightforward initialisation of the correct driver / conversion factors.

| Sensor \\/, Word 0 bit &rightarrow; | 11 | 10 | 9 | 8 | 7 | 6 | 5 |
|---|---|---|---|---|---|---|---|
MS5837-30BA26 | 0 | 0 | 1 | 1 | 0 | 1 | 0
MS5837-02BA01 | 0 | 0 | 0 | 0 | 0 | 0 | 0
MS5837-02BA21 | 0 | 0 | 1 | 0 | 1 | 0 | 1

It would also be nice if the `librariesAP_Baro/MS5611.*` files can be renamed to MS5xxx, to better reflect the multitude of sensor drivers they actually support.

**Describe alternatives you've considered**
The driver switching could alternatively be done manually with a parameter, as was discussed somewhat [in a previous issue](https://github.com/ArduPilot/ardupilot/issues/8258#issuecomment-398595956) (at the time it was seemingly suggested/thought there that the differentiation information wasn't available/queryable from the sensors), but that's less convenient and could require some hacky workarounds for driver handling, which doesn't seem desirable or necessary.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[x] Submarine

**Additional context**
There's [a long forum thread](https://discuss.bluerobotics.com/t/ardusub-support-for-bar02-depth-pressure-sensor-depth-hold-not-functioning/4689/) about ArduSub support for this, with multiple people asking for it to be added.

This would also make the "water pressure sensor" idea more viable for boats, as requested in #20721.

Shaikimram, 2025-01-22:
hey @ES-Alexander, I have created a PR for this issue could you please review it

sanggusti, 2025-03-31:
I don't think this is a `good first issue`. The PR mentioned in this issue is complicated and didn't even closed after several conversations and approvals.

ES-Alexander, 2025-03-31:
Fair point @sanggusti - it still seems possible using one of the approaches discussed [here](https://github.com/ArduPilot/ardupilot/pull/29122#issuecomment-2648486046), but I agree that it's no longer as straightforward as it initially seemed. It likely shouldn't be considered a good first issue anymore, at least until a detection approach is decided upon and more properly investigated.

PR #29122 (2025-06-26): AP_Baro: Support MS5837-02BA #27787
The changes include differentiation between multiple MS5xxx sensor variants using the PROM Word 0 bits [11:5], enabling automatic initialization and configuration of the correct driver. Additionally, this update includes specific calculations and second-order compensation logic for the MS5837-02BA sensor, making it more viable for shallow water applications.

Key Changes
Differentiation Logic:

Extracted product ID from PROM Word 0 bits [11:5] to identify sensor variants:
MS5837-30BA26: 0b0011010
MS5837-02BA01: 0b0000000
MS5837-02BA21: 0b0010101
New Sensor T [... cut]