ArduPilot/ardupilot #27795: SpeedyBeeF405WING building fail

IvanLeu, 2024-08-08:
I've run into issues when trying to run sitl_on_hw.py script for SpeedyBeeF405WING, but after all files compiled i get this error message: /opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld:common.ld:202 cannot move location counter backwards (from 00000000081069e0 to 0000000008100000)
collect2: error: ld returned 1 exit status

Waf: Leaving directory `/home/autopilot2024/arduplane-test/ardupilot/build/SpeedyBeeF405WING'
Build failed
 -> task in 'bin/arduplane' failed (exit status 1):
        {task 140552478781088: cxxprogram AP_Arming.cpp.52.o,AP_ExternalControl_Plane.cpp.52.o,ArduPlane.cpp.52.o,Attitude.cpp.52.o,GCS_Mavlink.cpp.52.o,GCS_Plane.cpp.52.o,Log.cpp.52.o,Parameters.cpp.52.o,Plane.cpp.52.o,RC_Channel.cpp.52.o,VTOL_Assist.cpp.52.o,afs_plane.cpp.52.o,altitude.cpp.52.o,avoidance_adsb.cpp.52.o,commands.cpp.52.o,commands_logic.cpp.52.o,control_modes.cpp.52.o,ekf_check.cpp.52.o,events.cpp.52.o,failsafe.cpp.52.o,fence.cpp.52.o,is_flying.cpp.52.o,mode.cpp.52.o,mode_LoiterAltQLand.cpp.52.o,mode_acro.cpp.52.o,mode_auto.cpp.52.o,mode_autotune.cpp.52.o,mode_avoidADSB.cpp.52.o,mode_circle.cpp.52.o,mode_cruise.cpp.52.o,mode_fbwa.cpp.52.o,mode_fbwb.cpp.52.o,mode_guided.cpp.52.o,mode_loiter.cpp.52.o,mode_manual.cpp.52.o,mode_qacro.cpp.52.o,mode_qautotune.cpp.52.o,mode_qhover.cpp.52.o,mode_qland.cpp.52.o,mode_qloiter.cpp.52.o,mode_qrtl.cpp.52.o,mode_qstabilize.cpp.52.o,mode_rtl.cpp.52.o,mode_stabilize.cpp.52.o,mode_takeoff.cpp.52.o,mode_thermal.cpp.52.o,mode_training.cpp.52.o,motor_test.cpp.52.o,navigation.cpp.52.o,parachute.cpp.52.o,qautotune.cpp.52.o,quadplane.cpp.52.o,radio.cpp.52.o,reverse_thrust.cpp.52.o,sensors.cpp.52.o,servos.cpp.52.o,soaring.cpp.52.o,system.cpp.52.o,tailsitter.cpp.52.o,takeoff.cpp.52.o,tiltrotor.cpp.52.o,tuning.cpp.52.o -> arduplane}
 (run with -v to display more information). 
 
 However on other boards it runs perfectly. How do I address this issue?

IamPete1, 2024-08-08:
You ran out of flash. You will need to turn off some other features to make room.

closed 2024-08-08 by IamPete1

IvanLeu, 2024-08-08:
> You ran out of flash. You will need to turn off some other features to make room.

I'm quite new to this, could you provide more details how to do it? Which features I can turn off and how to do it?

IamPete1, 2024-08-08:
If you don't need it turning off quadplane will save a lot. Add `define HAL_QUADPLANE_ENABLED 0` to the hwdef for the board. The developer discord is a great place for such questions. https://discord.gg/X54UyG6W