ArduPilot/ardupilot #27846: QuadPlane Frame Class Deca Not Supported

missionimaa, 2024-08-15:
## Bug report

**Issue details**
Looking at the full parameter list for Plane, Q_Frame_Class = 14 (Deca) is a valid option and going over the AP_MotorsMatrix, Deca seems to be supported for types 'Plus' and 'X'. 
However when I configure my AP's frame class to be Deca, I get messages saying "Config Error: fix problem then reboot" and "Config Error: Unsupported Q_Frame_Class 14".

**Version**
Arduplane 4.5.0

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadplane - Fixed wing with 10 hover motors and a throttle pusher. 

**Hardware type**
Cube Orange

IamPete1, 2024-08-15:
It just needs adding to this switch statement. 

https://github.com/ArduPilot/ardupilot/blob/61bd5e9bae1e64c6a216f2a0c937a61053955670/ArduPlane/quadplane.cpp#L684

IamPete1, 2024-08-21:
Closed by https://github.com/ArduPilot/ardupilot/pull/27874

closed 2024-08-21 by IamPete1